agherzan/meta-raspberrypi #756: VC4 driver not displaying output on older Pi varients.

buxtonpaul, 2020-10-24:
**Description**
On A Raspberry Pi or a Raspberry Pi Zero W, when the kernel reaches the point of enabling the VC4 driver, the display turns off. If VC4 support is disabled by editing config.txt then the boot process works fine and I get a terminal console. 
In either case the machines are alive and can be connected to via ssh.

Looking at dmesg it looked like the driver is started without any obvious problem.

Using a Raspberry Pi3 works fine.

The build is a very basic core-image-minimal the only additional thing is some postbuild scripts to configure wpa_supplicant for my network.

I am using the dunfell branch.

bblayers.
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/paulbuxton/piyocto/poky/meta \
  /home/paulbuxton/piyocto/poky/meta-poky \
  /home/paulbuxton/piyocto/poky/meta-yocto-bsp \
  /home/paulbuxton/piyocto/meta-raspberrypi \
  /home/paulbuxton/piyocto/meta-wifi-credentials/sysvinit \
  "

interesting bits of local.conf
# CONF_VERSION is increased each time build/conf/ changes incompatibly and is used to
# track the version of this file when it was generated. This can safely be ignored if
# this doesn't mean anything to you.
CONF_VERSION = "1"
IMAGE_FSTYPES ="rpi-sdimg"

CORE_IMAGE_EXTRA_INSTALL += " dropbear"

# MACHINE ??= "raspberrypi"
# MACHINE ??= "raspberrypi3-64"
MACHINE ??= "raspberrypi0-wifi"

Is the VC4 driver expected to work on older Pi's? Is there some additional configuration required for the image? 

Thanks,
Paul

agherzan, 2020-10-27:
Can you cross-check it with raspbian? I haven't tested older RPis in a while now.

buxtonpaul, 2020-10-27:
So tested a fresh Raspian image (from the pi imager  app), on 
Pi 1
Pi 2
Pi 0
Pi 3
and it works ok.
Modified config.txt to uncomment the vc4 fkms and tested again on Pi2, Pi0. Still working.
Then used raspi-config to try both Fake KMS and Fulll KMS varients. Both of those still working. 
So could be either version difference between the version used by Yocto vs Raspian, or could be some configuration.

agherzan, 2020-10-27:
Can you try to switch the kernel in Yocto the exact version raspbian has and give it a spin?

buxtonpaul, 2020-10-27:
Sure,
 I just need to figure out how. I am just getting started with Yocto :-)
The rasbpian kernel version is 5.4.72-v7+ soI am trying specifying 
PREFERRED_VERSION_linux-yocto ="5.4.72-v7+" in my local.conf. But it looks to be suspiciously quick at rebuilding it so I may need to dig in some more to figure out the correct way to swap kernel version.

agherzan, 2020-10-27:
You will need to switch this revision: https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi_5.4.bb#L4

buxtonpaul, 2020-10-29:
Well, this is embarrasing. I now seem to be unable to reproduce this problem either with the original kernel or the raspian one. Between reporting the issue and now I did trash an sd card I was using, so I am going to blame that.
I do have a different, but similar issue related to the VC4-kms driver/overlay, but I will open a new issue for that.

closed 2020-10-29 by buxtonpaul