agherzan/meta-raspberrypi #759: Cannot boot RPI4 using uboot on branch dumfell.

HenryGiraldo, 2020-11-03:
Cannot boot RPI4 using uboot on branch dumfell.

```
Found U-Boot script /boot.scr
422 bytes read in 25 ms (15.6 KiB/s)
## Executing script at 02400000
Saving Environment to FAT... OK
Booting from mmcblk0p1
Wrong Image Format for bootm command
ERROR: can't get kernel image!
SCRIPT FAILED: continuing...
47516 bytes read in 35 ms (1.3 MiB/s)
Card did not respond to voltage select!
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-bcm283x-rpi
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-bcm283x
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
No ethernet found.
No ethernet found.
```

**Steps to reproduce the issue:**
Build yocto/poky using meta-raspberry dumfell branch with following configuration in local.conf:
```
RPI_USE_U_BOOT = "1"
ENABLE_UART= "0"
IMAGE_FSTYPES = "rpi-sdimg ext4.gz"
```

Layers:
```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /workdir/layers/poky/meta \
  /workdir/layers/poky/meta-poky \
  /workdir/layers/poky/meta-yocto-bsp \
  /workdir/layers/meta-openembedded/meta-oe \
  /workdir/layers/meta-openembedded/meta-multimedia \
  /workdir/layers/meta-openembedded/meta-python \
  /workdir/layers/meta-openembedded/meta-networking \
  /workdir/layers/meta-raspberrypi \
  /workdir/layers/meta-swupdate \
  /workdir/layers/meta-swupdate-boards \
  "
```

core-image-full-cmdline-raspberrypi4-20201103115021.rootfs.rpi-sdimg flashed into SD card.

I've tried with MACHINE=raspberrypi4-64 and it fails as well.

agherzan, 2020-11-03:
Please try it out without any extra layers. Just poky, meta-oe and meta-raspberrypi and tell me if it works.

HenryGiraldo, 2020-11-03:
Without extra layers:
```
BBLAYERS ?= " \
  /workdir/layers/poky/meta \
  /workdir/layers/poky/meta-poky \
  /workdir/layers/meta-openembedded/meta-oe \
  /workdir/layers/meta-raspberrypi \
  "
```

Goes further, stays in Starting kernel...

```
U-Boot 2020.01 (Jan 06 2020 - 20:56:31 +0000)

DRAM:  3.9 GiB
RPI 4 Model B (0xc03111)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... *** Warning - bad CRC, using default environment

In:    serial
Out:   serial
Err:   serial
Net:   Net Initialization Skipped
No ethernet found.
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found U-Boot script /boot.scr
263 bytes read in 26 ms (9.8 KiB/s)
## Executing script at 02400000
6263200 bytes read in 1225 ms (4.9 MiB/s)
Saving Environment to FAT... OK
## Booting kernel from Legacy Image at 00080000 ...
   Image Name:   Linux-5.4.59-v7l
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    6263136 Bytes = 6 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 2eff4100
   Booting using the fdt blob at 0x2eff4100
   Loading Kernel Image
   Using Device Tree in place at 2eff4100, end 2f002fe5

Starting kernel ...
```

agherzan, 2020-11-03:
It looks a duplication of https://github.com/agherzan/meta-raspberrypi/issues/732

agherzan, 2020-11-03:
There is also a fix there but we need more info from the author.

closed 2020-11-03 by agherzan