agherzan/meta-raspberrypi #765: unable to boot raspberry pi 3b using 64 bit configuration

AnanduKE, 2020-11-20:
Hi,
I am unable to boot raspberry pi 3b using 64 bit configuration. Boot just hangs after
 Starting syslogd/klogd: done
i am using dunfell branch.
Enabled uart and currently using uart for getting console

agherzan, 2020-11-20:
Can you paste UART boot logs and also build configuration?

abhi-84, 2021-01-06:
I am facing the similar issue but I am not using UART. I ave tried with both dunfell and gatesgarth but same issue. With Warrior branch it is working fine. I wanted to access pi using ssh but ethernet port is not getting activated.
When connected with HDMI, I am getting following message:
[https://i.stack.imgur.com/dpaET.jpg](url)
Please help

Abhishek

agherzan, 2021-01-07:
That is not the same issue. Your boot fails while loading the kernel image in uboot while this issue reports hanging in userspace. As there is no update on the initial issue, I'll close this.

closed 2021-01-07 by agherzan