agherzan/meta-raspberrypi #766: u-boot fails to build for raspberrypi3-64 and raspberrypi4-64

jordemort, 2020-11-25:
I am attempting to build an image for raspberrypi4-64 that uses u-boot. It consistently fails to build, apparently because it is unable to find libgcc:

```
|   aarch64-bantha-linux-ld.bfd    -Ttext 0xc100000 -g -o examples/standalone/hello_world -e hello_world examples/standalone/hello_world.o examples/standalone/libstubs.o -L . -lgcc
| aarch64-bantha-linux-ld.bfd: cannot find -lgcc
```

If I try switching MACHINE to raspberrypi4 or raspberrypi3 or qemux86-64 it builds fine. raspberrypi3-64 fails for me in the same way that raspberrypi4-64 does.

From what I can see, it seems like this is a fairly common problem when building u-boot in general, but I did not see a clear solution. It sounds like other people are able to build u-boot for this platform successfully, so it is possible (likely?) that there is something in my layer configuration that is messing things up.

**Steps to reproduce the issue:**
1. Assemble a project using the `dunfell` branches of poky, meta-raspberrypi, meta-updater, and meta-updater-raspberrypi
2. Set MACHINE to raspberrypi4-64
3. bitbake -v u-boot

**Describe the results you received:**

u-boot fails to compile, because it can't find libgcc

**Describe the results you expected:**

u-boot compiles successfully

**Additional information you deem important (e.g. issue happens only occasionally):**

This seems to happen consistently

**Additional details (revisions used, host distro, etc.):**

```
meta                 
meta-poky            
meta-yocto-bsp       = "dunfell:235dff82276a8a2ff4d0ee1cf93243925537551c"
meta-oe              
meta-python          
meta-networking      
meta-filesystems     = "dunfell:f2d02cb71eaff8eb285a1997b30be52486c160ae"
meta-raspberrypi     = "dunfell:987993209716302eb8f314f69a2a3340555f94d8"
meta-updater         = "dunfell:cc1b38310f4ebd0558438057a1edfeb4a3555b16"
meta-updater-qemux86-64 = "dunfell:71e2e2426b09e5aed6eabed72c8d289c0343c2de"
meta-updater-raspberrypi = "dunfell:3e4795b85861e63c54f6f7c573f8bd13b3024072"
meta-virtualization  = "dunfell:ff997b6b3ba800978546098ab3cdaa113b6695e1"
meta-bantha          = "meta-updater:9a01bec6f8e3483ad645bda05b56ac7f0f4490c4"
```

Host distro: I'm using the crops/poky:debian-10 container

kraj, 2020-11-25:
Can you add `DEPENDS_append_raspberrypi-64 = " libgcc"` in `u-boot_%.bbappend` in your layer

jordemort, 2020-11-25:
@kraj I did try adding a libgcc dependency but it didn't fix it - it seems to already have the dependency, at least transitively.

I went searching through my layers to find things that were fiddling with u-boot and masking them out; I think the culprit was [this bbappend from meta-updater-qemux86-64](https://github.com/advancedtelematic/meta-updater-qemux86-64/blob/master/recipes-bsp/u-boot/u-boot_%25.bbappend) - after adding that to BBMASK I was able to build u-boot successfully.

I'll leave this open for a bit in case there are some suggestions on how to fix that but I'll close it after I get started with work tomorrow, it looks like this is a meta-updater-qemux86-64 problem and not a meta-raspberrypi problem.

kraj, 2020-11-25:
Change
```
SRC_URI +=" \
            file://0001-Set-up-environment-for-OSTree-integration.patch \
            file://0001-Revert-ide-remove-duplicate-defines-form-include-ata.patch \
          "

# fix after default security flags in poky
TOOLCHAIN_OPTIONS_append = "${SECURITY_NOPIE_CFLAGS}"
```
to

```
SRC_URI_append_qemux86-64 =" \
            file://0001-Set-up-environment-for-OSTree-integration.patch \
            file://0001-Revert-ide-remove-duplicate-defines-form-include-ata.patch \
          "

# fix after default security flags in poky
TOOLCHAIN_OPTIONS_append_qemux86-64 = " ${SECURITY_NOPIE_CFLAGS}"
```

that should help

jordemort, 2020-11-25:
@kraj Thanks! I've opened an issue over at the other layer's repo, I think I can go ahead and close this now.

closed 2020-11-25 by jordemort