agherzan/meta-raspberrypi #772: Using custom machine name -> linux-raspberrypi issues

nandra, 2020-11-30:
I'm adding new machine based on raspberrypi3. So in my meta-layer I added `my-super-machine.conf` with content:
```
require conf/machine/raspberrypi3.conf
```

I'm hitting issue during the build that linux-raspberrypi 5.4 (I'm on dunfell release) report:
```
ERROR: linux-raspberrypi-1_5.4.72+gitAUTOINC+5d52d9eea9_154de7bbd5-r0 do_kernel_metadata: Could not locate BSP definition for my-super-machine/standard and no defconfig was provided
```

I tried to set ```KMACHINE="raspberrypi3"``` but it still says:
```
ERROR: Could not locate BSP definition for raspberrypi3/standard and no defconfig was provided
```

Ideas how to fix this issue? Thanks.

pino-kim, 2020-11-30:
Hello? @nandra 
I just read your issue and share my idea.

I make qemuarm64 machine for the test, I set my defconfig like bellow. 
https://github.com/pino-kim/meta-kernel-dev/blob/master/conf/machine/qemuarm64-kdev.conf

I think you miss KBUILD_DEFCONFIG parameter in your conf file.

cheer.

nandra, 2020-11-30:
@pino-kim thanks a lot for for pointer. Setting value resolve my issue :+1:

closed 2020-11-30 by nandra

agherzan, 2020-11-30:
Yes. It is the defconfig definition which is done on a per-machine basis.

nandra, 2020-11-30:
Hmm I found 2 ways how to resolve an issue:
1. Add to machine,conf ```MACHINEOVERRIDES .= ":raspberrypi3"````
2. Add linux-raspberrypi_%.bbappend with : ```KBUILD_DEFCONFIG_my-super-rpi-machine = "bcm2709_defconfig"```

Is there pros/cons using 1 or 2? Thanks.

agherzan, 2020-11-30:
It very much depends on what your machine sw requirements are. If you currently have a hw platform which needs custom defconfig and/or is significantly different from all current machines, use 2. Otherwise use 1 and take advantage of all the rpi3 changes across the layer.