agherzan/meta-raspberrypi #781: RP-4 B image creation seems to have failed...

arunlee77, 2020-12-16:
**Description**

HW: Raspberry Pi 4 B
SW: core-image-base created using kas as mentioned in the README
Build: Suceded, with no errors.

**Steps to reproduce the issue:**
1. clone this repo
2. and run kas build meta-raspberrypi/kas-poky-rpi.yml
3. Output of images folder 

~/work/yocto/build/tmp/deploy/images/raspberrypi4$ ls *.bz2
core-image-base-raspberrypi4-20201216125949.rootfs.tar.bz2  core-image-base-raspberrypi4.tar.bz2
core-image-base-raspberrypi4-20201216125949.rootfs.wic.bz2  core-image-base-raspberrypi4.wic.bz2
 
4. Is this the expected outcome of the build?
5. I copied the *.wic.bz2 to a 16GB SD Card using Balena Etcher, but that doesn't boot.

I am doing something totally wrong here?
Do i need any other additional steps to get a valid image which i can use to boot RP4?

kraj, 2020-12-16:
did you create the SD card image, when you mount it on your PC what is the content of the SD card.

arunlee77, 2020-12-16:
> 
> 
> did you create the SD card image, when you mount it on your PC what is the content of the SD card.

There is no sdimg created from dunfell branch

arunlee77, 2020-12-17:
I re created the yocto environment from scratch and built the core-image-base.
It looks like the image works only with bmaptool and not with Etcher.

I am closing this issue, thank you for your directions.

closed 2020-12-17 by arunlee77