agherzan/meta-raspberrypi #786: overwrite splash screen

bernardoaraujor, 2020-12-21:
**Description**

I'm trying to overwrite the RPi splash screen.
Followed the instructions from https://stackoverflow.com/a/63732589
no success.

**Steps to reproduce the issue:**
1. `psplash_%.bbappend:`
```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
SPLASH_IMAGES_append = " file://foo-img.png;outsuffix=bar"
ALTERNATIVE_PRIORITY_psplash-bar[psplash] = "200"
```
2. `conf/local.conf:`
```
PACKAGE_INSTALL_append = " psplash-bar"
```

**Describe the results you received:**
Still shows Raspberry Pi logo on splash screen.

**Describe the results you expected:**
Custom image on splash screen.

**Additional details (revisions used, host distro, etc.):**
Tested under dunfell.

gvkalra, 2020-12-22:
I am under dunfell as well, and the following works for me:

append recipe:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"
SPLASH_IMAGES_seesv = "file://psplash-seesv-img.h;outsuffix=seesv"
```

image recipe:
```
SPLASH = "psplash-seesv"

```

**Note:** seesv is my own `BBFILE_COLLECTIONS` with `BBFILE_PRIORITY_seesv = "30"`

agherzan, 2020-12-24:
Thanks, @gvkalra . @bernardoaraujor , good to close?

bernardoaraujor, 2020-12-24:
thanks @gvkalra your solution only differs from mine with regards to `BBFILE_PRIORITY`.
I set mine to `30` like yours and I still see the RPi logo during boot.

bernardoaraujor, 2020-12-24:
actually my bad! I was doing `SPLASH_IMAGES_append` instead of `SPLASH_IMAGES_bar`.
works fine now.

sorry @agherzan
thanks again @gvkalra 

closing.

closed 2020-12-24 by bernardoaraujor

agherzan, 2020-12-24:
That's absolutely fine. Glad it's sorted.