agherzan/meta-raspberrypi #789: Boot Log does not show

embexus, 2021-01-09:
Boot Log does not show on raspberrypi3 with LCD 7'' display

**Steps to reproduce the issue:**
1. Build meta-raspberrypi (Gatesgarth) default settings

**Describe the results you received:**

No Logo
**Describe the results you expected:**

Logo with 4x raspberries

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
poky gatesgarth

embexus, 2021-01-09:
there should be no coupling between loglovel (quiet) and boot logo display

embexus, 2021-01-09:
it seems that this Bug is a feature in linux raspberrypi kernel !

https://github.com/raspberrypi/linux/commit/10993504d647356196a04b3022d645ec92e00159

agherzan, 2021-01-12:
@embexus Thanks for reporting this. I'll defer this to the linux raspberrypi project. For me this upstream change makes sense but if for your specific usecase it doesn't, probably the easiest way forward would be a revert in a local layer. Much more elegant would be to take it upstream and maybe implement it as an argument.

closed 2021-01-12 by agherzan