agherzan/meta-raspberrypi #790: Build fails due to gstreamer

deep5050, 2021-01-13:
**Description**
```
ERROR: No recipes in default available for:
  /home/dipankar/poky/meta-raspberrypi/recipes-multimedia/gstreamer/gstreamer1.0-omx_1.18%.bbappend
  /home/dipankar/poky/meta-raspberrypi/recipes-multimedia/gstreamer/gstreamer1.0-plugins-good_1.18.%.bbappend
```
Showing this error while building `core-image-base`

**Steps to reproduce the issue:**
1. `bitbake core-image-base`
2.
3.

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
host: Ubuntu 16.04

kraj, 2021-01-13:
are you using matching branch of meta-raspberrypi with poky/

deep5050, 2021-01-13:
> are you using matching branch of meta-raspberrypi with poky/

using `master`

kraj, 2021-01-13:
in `local.conf` add
```
LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx "
```

deep5050, 2021-01-13:
> in `local.conf` add
> 
> ```
> LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx "
> ```

thanks for the quick response, but this did not help me

deep5050, 2021-01-13:
> > in `local.conf` add
> > ```
> > LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx "
> > ```
> 
> thanks for the quick response, but this did not help me

Even I switched to `gatesgarth` branch both for poky and meta-raspberry. still having this issue.

deep5050, 2021-01-13:
i had to add gstreamer layer manually to resolve it.

closed 2021-01-14 by deep5050