agherzan/meta-raspberrypi #793: Pi 400 and CM4 support

shawaj, 2021-01-15:
**Description**

Add overlays for pi 400 and cm4 support.

shawaj, 2021-01-15:
Added a commit here https://github.com/agherzan/meta-raspberrypi/pull/783/commits/8c931c3782bbabb83cca80b70fdb368601869ecf

agherzan, 2021-01-15:
`This commit does not belong to any branch on this repository, and may belong to a fork outside of the repository.` Can you raise a PR? have you tested that all the kernel versions we currently have in the layer have those dts? files

agherzan, 2021-01-15:
You meant to mention https://github.com/agherzan/meta-raspberrypi/pull/783/commits

shawaj, 2021-01-15:
@agherzan my bad yes

immiz, 2021-02-15:
since https://github.com/agherzan/meta-raspberrypi/pull/783 is merged - this issue could be closed, right?

agherzan, 2021-02-15:
Cheers @immiz . Clsoed.

closed 2021-02-15 by agherzan

pabigot, 2021-11-18:
@agherzan @kraj I'm reopening this, though perhaps I'm wrong and it should be a new issue: the issue title is broader than its original description.

#793 was to add support for CM4 and Pi 400.  This was closed when #783 was merged, but https://github.com/agherzan/meta-raspberrypi/pull/783#issuecomment-749805660 has:

> * CM4 - we have no BSP for this yet as far as I am aware. But I guess we can do a similar thing with the Pi 4? The CM4 and Pi400 should all work using the Pi4 base image. We just need to pull in the new wifi drivers I believe as they both use the BCM43456 not the BCM43455 which is on the Pi 4
> 
> Based on this - what do you want to do? Merge PR and then remove CM3/CM1 specific BSPs and then amend the Pi4 one to support Pi400 and CM4?

AFAICT some of these steps got dropped: there's nothing that will ensure `linux-firmware-rpidistro-bcm43456` is installed:
```
2021-08-06T14:41:10.125587+00:00 raspi400 kernel: [    7.003078] brcmfmac: F1 signature read @0x18000000=0x15294345
2021-08-06T14:41:10.125590+00:00 raspi400 kernel: [    7.123970] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43456-sdio for chip BCM4345/9
2021-08-06T14:41:10.125594+00:00 raspi400 kernel: [    7.143523] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43456-sdio.bin failed with error -2
```

I've confirmed that manual installation of that package allows Wi-Fi to work on on the Pi 400.

I don't know how to add a CM4/Pi400 BSP for this, or whether it's necessary.  Would to adding linux-firmware-rpidistro-bcm43456 to `MACHINE_EXTRA_RRRECOMMENDS` for raspberrypi4-* machines be the right solution?

kraj, 2021-11-18:
I guess
Adding extra firmware to pi4 machine class is ok they have enough disk and firmware size is not huge

pabigot, 2021-11-18:
OK, thanks, I'll do a PR for that.