agherzan/meta-raspberrypi #794: use alternative kernel config

embexus, 2021-01-18:
Hi I would like to use an alternative defconfig for rpi3

I created in my layer : my-layer/recipes-kernel/linux/linux-raspberrypi_%.bbappend:

```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI += " file://defconfig"
```
defconfig is a minimalist config and was generated using:

```
MACHINE=raspberrypi3 bitbake -c menuconfig virtual/kernel
```

it seems after then that my defconfig is ignored.

what is the best way to use another kernel config ?

embexus, 2021-01-18:
In Build logs I get:
```
WARNING: linux-raspberrypi-1_5.4.72+gitAUTOINC+5d52d9eea9_154de7bbd5-r0 do_kernel_configme: [NOTE]: defconfig was supplied both via KBUILD_DEFCONFIG and SRC_URI. Dropping SRC_URI defconfig
```

kraj, 2021-01-18:
you might have to set
```
KBUILD_DEFCONFIG = "defconfig"
```
in your bbappend for kernel recipe.

agherzan, 2021-01-19:
You will need to check it - as it has been a while since I played with these - but `KBUILD_DEFCONFIG` is meant for in-kernel sources defconfig. If you want an external one, you should be able to do it by adding it to `SRC_URI` and nuking `KBUILD_DEFCONFIG`? That's also what the error seems to imply.

embexus, 2021-01-19:
@kraj this does not seems to have any effect :-(

could it be because of the following:

`KCONFIG_MODE = "--alldefconfig"` 

https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi.inc#L19

embexus, 2021-01-19:
@agherzan I'm not using external Kernel sources, just another defconfig and I would like to have a lightweight kernel configuration

agherzan, 2021-01-19:
External defconfig, not external sources.

kubasejdak, 2021-01-25:
I had a similar problem. Even more, I'm unable to even provide a patch for the existing one (bcm2711_defconfig).

I did something like this in `<my-layer>/recipes-kernel/linux/linux-raspberrypi_5.4.bbappend`:

```
FILESEXTRAPATHS_prepend := "${THISDIR}/files:"

SRC_URI += "file://001-custom-kernel-version.patch"
```

The patch was done with the following sequence of commands:
```
bitbake virtual/kernel -c devshell
bitbake ARCH=arm bcm2711_deconfig
bitbake menuconfig # change only local version
bitbake savedefconfig
git diff > <path_to_my_patch>
```

After compilation kernel had still old local version. Why is this not working?

de14a447, 2021-05-16:
@embexus did you ever get this solved?

embexus, 2021-05-16:
@de14a447 : here my workarround:

```
SRC_URI += " file://myrpi3_defconfig"

addtask copy_defconfig after do_kernel_configcheck before do_compile

do_copy_defconfig () {

    cp ${WORKDIR}/myrpi3_defconfig ${B}/.config
}
```

emilm, 2021-07-28:
https://github.com/agherzan/meta-raspberrypi/blob/dunfell/recipes-kernel/linux/linux-raspberrypi.inc#L87
This might be the culprit. It should be enough to do what @embexus  did, or simply:

```
unset KBUILD_DEFCONFIG

do_configure_prepend () {
	cp "${WORKDIR}/defconfig" "${B}/.config"  
}
```

closed 2021-10-30 by agherzan

amamory, 2022-01-13:
For future references, [here](https://github.com/amamory-embedded/meta-myrpi/tree/main/recipes-kernel/linux/linux-raspberrypi) is an example of a yocto layer built on top of  meta-raspberrypi that creates new images and applies configurations to the kernel. 

It applies a complete defconfig file and also a kernel fragments (the commented code in the bbappend file). Example working with `dunfell`.  Check out the readme for further instruction.

Cheers!

djr-spectrummfg, 2023-10-05:
The above solutions no longer work, at least not with devtool. I'm using the kirkstone branch.