agherzan/meta-raspberrypi #800: Not getting full 4 cores usage on Pi 4

lorenzo-arena, 2021-01-22:
I am trying to do some power measures on with a Yocto build based on the dunfell branch, on a Raspberry Pi 4. Running a bunch of `yes > /dev/null &` to stress the four cores and using `top` to monitor them, I expected to see the usage go near 400%, instead I only managed to get about 100%. From the `/sys/devices/system/cpu/online` file all 4 cores seems to be online.
Is this a known issue?
Under Raspbian running the same commands I get the foreseen 400% usage.

lorenzo-arena, 2021-01-22:
Seems like busybox `top` shows 100% as the maximum; using `htop` show that all four cores are at 100% usage.

closed 2021-01-22 by lorenzo-arena

kraj, 2021-01-22:
Right use top from procps package