agherzan/meta-raspberrypi #802: KMS is not working on Raspberry PI 4 / 64bit / vc4-kms-v3d

v-morlock, 2021-01-27:
**Description**

KMS is not working with yocto/dunfell on 64bit Raspberrypi 4 using the vc4-kms-v32-Driver. When I'm using the fkms-driver, everything works fine. 

**Steps to reproduce the issue:**
1. build a core-image-minimal with those settings on dunfell:
```
MACHINE = "raspberrypi4"
VC4DTBO = "vc4-kms-v3d"
DISABLE_OVERSCAN = "1"
ENABLE_UART = "1"
IMAGE_INSTALL_append = " kmscube"
```

2. run kmscube

**Describe the results you received:**
```
drmGetDevices2 failed: No such file or directory
could not open drm device
failed to initialize legacy DRM
```
**Describe the results you expected:**
kmscube working.

Any help appreciated!

matt-hammond-001, 2021-06-15:
Have you tried ```VC4DTBO = "vc4-kms-v3d-pi4"``` ?

agherzan, 2021-10-30:
That is expected. The magic relies in overlay_map.dtb in later kernels. This will dynamically load the dtbo @matt-hammond-001 is mentioning above. I know we have an older kernel on dunfell so you'll have to check for the required support.

closed 2021-11-14 by agherzan

bartiszosti, 2024-08-06:
Hello,
I have same issue on the scarthgap branch.
VC4DTBO = "vc4-fkms-v3d" works but VC4DTBO = "vc4-kms-v3d" doesn't work.
As well I have terrible GUI performance, laggy mouse etc.