agherzan/meta-raspberrypi #805: Add kernel 5.10

krasi-georgiev, 2021-02-02:
Hi. I was advised by the balena forum that in order to get the 5.10 kernel for the RPI4 need to requests it here.
https://forums.balena.io/t/can-you-upgrade-the-kernel-to-5-10/245763/13

I am using the JamHat with the new improved GPIO Interface and to use the button debounce(required due to the JamHat's electrically noisy switch) it requires kernel 5.10

Here are more details on why I need to use the debounce
https://github.com/warthog618/gpiod/issues/13

And here is a video for the new GPIO kernel features
https://youtu.be/cdTLewJCL1Y

agherzan, 2021-02-03:
You already have a 5.10 version included. Check https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi-dev.bb

krasi-georgiev, 2021-02-04:
Thanks for the update. could you give some estimation on when would it be promoted to a stable release as according to reply from Michal Toman on the balena forum only then  can be use for the balena OS
https://forums.balena.io/t/can-you-upgrade-the-kernel-to-5-10/245763/30?u=krasi-georgiev

agherzan, 2021-02-05:
I see. Most probably that will happen sometime next week. I'll reopen this to have it tracked.

krasi-georgiev, 2021-02-05:
Ohh, nice magic if this gets in so soon!

agherzan, 2021-02-06:
Added as part of https://github.com/agherzan/meta-raspberrypi/pull/808

closed 2021-02-06 by agherzan