agherzan/meta-raspberrypi #809: Mount before systemd starts

om26er, 2021-02-05:
This isn't really a bug, rather a question (not sure if this is the right place for that).

We have a need to mount a few mount points before SystemD starts. What would be the right place to do that ? From what I understand, by default there is no initramfs created for RaspberryPi, so where should the change really live ?

kraj, 2021-02-06:
you need initramfs as you rightly said. Now you could try to enable it and the knobs are well documented [here](https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#image-with-initramfs) but you need to define logic. In yoe distro we use initramfs based boot system by default e.g. see

https://github.com/YoeDistro/yoe-distro/tree/master/sources/meta-yoe/recipes-support/updater

basically yoe-simple-image will automatically boot using this updater/initramfs and now you can encode the mount logic in the script on launch see

https://github.com/YoeDistro/yoe-distro/blob/master/sources/meta-yoe/recipes-support/updater/files/updater.installer

agherzan, 2021-02-06:
If you only need a set of mounts you can also code it as part of a wrapper which at the end execs the init process. That would be less overhead.

Anyway, I'll close this issue for now.

closed 2021-02-06 by agherzan

om26er, 2021-02-06:
> If you only need a set of mounts you can also code it as part of a wrapper which at the end execs the init process. That would be less overhead.

Yeah, I only need to do mounts. Where exactly shall I make the change ?  If getting away with using initramfs is an option, I'd go with that one.

agherzan, 2021-02-07:
In that case you can use what @kraj recommended or directly use he initramfs frameworkiin yocto and add a custom module.

om26er, 2021-02-07:
So I was thinking a bit more about it. I think for my case I can get away with just change the `/sbin/init` symlink to be a full fledged script that in the end runs `/lib/systemd/systemd`

agherzan, 2021-02-07:
That is exactly what I gave you as an option above. That should be easy to do and less overhead as a full-fledgediinitramfs .

om26er, 2021-02-07:
Ah, I had misunderstood that one. Thanks a lot :-)