agherzan/meta-raspberrypi #812: Add u-boot defconfig for cm3

nzottmann, 2021-02-15:
**Description**

Configuring `MACHINE ?= "raspberrypi-cm3"` in `/build/conf/local.conf` uses `rpi_3_32b_config` in https://github.com/agherzan/meta-raspberrypi/blob/f5fe02c30cc82e3f47c36de76d7b73d2482018bf/conf/machine/raspberrypi-cm3.conf#L10
which refers to the defconfig for the rpi 3b, as there is no defconfig for the cm3 in u-boot.

The defconfig `rpi_3_defconfig` at https://github.com/u-boot/u-boot/blob/master/configs/rpi_3_defconfig uses the device tree for the rpi 3b:
```
CONFIG_DEFAULT_DEVICE_TREE="bcm2837-rpi-3-b"
```
which is located at https://github.com/u-boot/u-boot/blob/master/arch/arm/dts/bcm2837-rpi-3-b.dts

**Consequences**
The defconfig and device tree configure on-board peripherals of the rpi 3b (wifi/bt module for example), which is not necessarily present on a cm3 baseboard. This can lead to conflicts when using the pins different than on a rpi 3b, for example like in #519 

**Fix proposal**
I would add a cm3 defconfig `rpi_cm3_config` to u-boot. It should refer to the device tree file for the cm3 base board https://github.com/u-boot/u-boot/blob/master/arch/arm/dts/bcm2837-rpi-cm3-io3.dts or to a new device tree file for a blank baseboard, removing the hdmi part.

I would submit a pull request, but I am no expert on this and have a few questions:
- Would this better be added to u-boot itself or by a bbappend in meta-raspberrypi?
- Would it be ok to copy `rpi_3_defconfig` and change the device tree file or would other changes to the defconfig be recommended? I am not familiar with all these config options.
- Is `bcm2837-rpi-cm3-io3.dts` ok or should there be a new device tree file?

agherzan, 2021-03-16:
@nzottmann This sounds like a contribution to push to uboot first and we can either get it through an upgrade (from poky/oe-core) or backport a patch when it lands. What do you think?

nzottmann, 2021-03-19:
That sounds like the right way to go, I will make a contribution to uboot.

agherzan, 2021-03-23:
Thanks @nzottmann and keep us updated.

agherzan, 2021-11-14:
@nzottmann do you have any news on this?

nzottmann, 2021-11-15:
Sorry, unfortunately not. As you probably know, patches to u-boot can not be supplied via github pull request but have to be sent to the mailing list following a set of rules: https://www.denx.de/wiki/U-Boot/Patches

I know where this historically comes from and that there are reasons to have such a process, but I am not used to this and for me it's too complicated to understand and follow all these instructions for just one small patch. This is not very beginner friendly, so I worry about people not being happy with my submission after wasting hours to understand all the rules

Thus I decided to not make a contribution. From my point of view, you can close this issue as it is not a problem of `meta-respberrypi`.

For anyone coming here with the same problem:
I fixed this by using [devtool](https://wiki.yoctoproject.org/wiki/TipsAndTricks/Patching_the_source_for_a_recipe) to create a patch to modify the source of `recipes-bsp/u-boot/u-boot` in this repo:
- Create new device tree `arch/arm/dts/bcm2837-rpi-cm3-custom.dts`
- Add custom device tree to Makefile `arch/arm/dts/Makefile`
- Change `CONFIG_DEFAULT_DEVICE_TREE` in `configs/rpi_3_32b_defconfig b/configs/rpi_3_32b_defconfig` to custom device tree

jsebrd, 2022-05-13:
@nzottmann Do you have any repo/source you can share showing your custom .dts for CM3?  I'd be happy to team with you on solving some issues as I have the same requirement to get u-boot working on CM3 for eventual swupdate intergration.