agherzan/meta-raspberrypi #813: Not able to import python tty module

bhargavthriler, 2021-02-18:
**Description**

Error while python `tty` module
How to enable this module in yocto build ?
I am not sure if this the correct place to raise an issue

**Describe the results you received:**
```
root@raspberrypi3-64:~# python3
Python 3.9.1 (default, Dec  7 2020, 22:33:43) 
[GCC 10.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tty
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ModuleNotFoundError: No module named 'tty'
>>> 
root@raspberrypi3-64:~# 
```

**Describe the results you expected:**
```
bhargav@bhargav-Lenovo-Legion-Y540-15IRH-PG0:~$ python3
Python 3.6.9 (default, Oct  8 2020, 12:12:24) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tty
>>> 
bhargav@bhargav-Lenovo-Legion-Y540-15IRH-PG0:~$ 
```

kraj, 2021-02-18:
Please add `python3-terminal` to your `IMAGE_INSTALL`

bhargavthriler, 2021-02-18:
> Please add `python3-terminal` to your `IMAGE_INSTALL`

Yes this worked
Thanks

closed 2021-02-18 by bhargavthriler