agherzan/meta-raspberrypi #814: I2S Output

moose-rivieh, 2021-02-19:
Is it possible to configure this layer to output I2S on RPI pins? Kind of like how I2C and UART options are supported. I didn't see any reference to I2S in the features documentation.

kraj, 2021-02-19:
`ENABLE_I2C = "1"` in local.conf might help

moose-rivieh, 2021-02-19:
To clarify, I am looking to output I2S signal and not I2C. There are two different protocols. I already have I2C enabled and using it for something else. Page 110 in the RPI4 processor's datasheet describes this communication bus (https://datasheets.raspberrypi.org/bcm2711/bcm2711-peripherals.pdf).

kraj, 2021-02-19:
i misread it :) yeag we do not have i2s consolidated like this. you perhaps need to add a kernel config fragment via SRC_URI

moose-rivieh, 2021-02-19:
Yeah, I found this I2S reference in the RPI config.txt file
"## i2s
"##     Set to "on" to enable the i2s interface
"##
"##     Default off.
"##
"#dtparam=i2s=off"

I am thinking of enabling the interface by adding the following line to my local.conf file:
` RPI_EXTRA_CONFIG = "dtparam=i2s=on"`

Although I don't know how to send a wave file to this interface. I guess I need to use some ALSA tool, but I have no audio manipulation experience in Linux. Any input would be appreciated?

moose-rivieh, 2021-02-19:
I also found this in the conf.txt:
"## audio
"##     Enable the onboard ALSA audio
"##     Default off.
"#dtparam=audio=off

I am guessing I need to enable it as well. I just need to figure out how to point the ALSA driver to use the I2S output and not the analog output.

agherzan, 2021-02-25:
@moose-rivieh I haven't played with i2s that much but I always recommend to take a look at Raspbian and see how they enable it.

agherzan, 2021-10-30:
@moose-rivieh any progress ok this?

closed 2021-11-14 by agherzan