agherzan/meta-raspberrypi #824: CM4 USB not working

ghuband, 2021-03-10:
**Description**

**Steps to reproduce the issue:**
1.  Followed the directions in the README for the Yocto build setup.  Set all repo branches to Dunfell.
2.  In local.conf
MACHINE = "raspberrypi4-64"
ENABLE_DWC2_HOST = "1"
3.  bitbake core-image-minimal-dev
4. Connected the CM4 IO board to the host computer with a micro-usb cable
5. Inserted the jumper at J2 to disable eMMC boot, power on board
6. sudo dd bs=1M if=core-image-minimal-dev-raspberrypi4-64.rpi-sdimg of=/dev/sdf
7. Power off the board and removed the micro-usb cable and the J2 jumper.  Insert ethernet cable.
8. Power on the board.
9. ssh into root
10. Plug mouse into USB2 port.

**Describe the results you received:**
Mouse did function correctly, no sensor light on mouse, no connection shown from "udevadm monitor".

**Describe the results you expected:**
Mouse functions correctly, connection shown in "udevadm monitor".

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

aurelian17, 2021-03-10:
@agherzan I can confirm that this works on a 32bit configuration. I tested on a CM4 with 4Gb of RAM without eMMC, with WiFi.

asuhler, 2021-03-10:
@agherzan I can confirm the issue on both dunfell and master branchs using 64 bit. Additionally, I am unable to successfully configure the UART ttl with `"ENABLE_UART=1"` in `local.conf` both on the CM4 and a 4gb RPi4. The resulting config.txt has the correct values for both options in the resulting builds, but the functionality is not there.  
  
**Steps to reproduce the issue:**  
**CM4:**  
1. Pull clean versions of poky/master and meta-raspberrypi/master
2. In "local.conf" add:
- `MACHINE ?= "raspberrypi4-64"`
- `ENABLE_UART="1"`
- `ENABLE_DWC2_HOST = "1"`
- `IMAGE_FSTYPES="rpi-sdimg"`
- `CORE_IMAGE_EXTRA_INSTALL+="openssh openssh-sshd openssh-sftp openssh-sftp-server"`
3. `bitbake core-image-minimal`
4. Set the j2 jumper for disable eMMC boot and plug in the micro-usb cable to my windows desktop
5. Connect the CM4 board to windows with the `rpiboot` utility
6. Use `win32diskimager` to flash the eMMC with my image
7. Power off the board, remove the jumper from J2 and unplug the micro-usb cable
8. Boot the board
9. Log in via SSH as root
10. Plug in my backlit USB keyboard
11. Connect a 3.3v TTL to USB converter to the UART on the GPIO header (J8)
12. Attempt connection to the UART ttl  

**RPi 4:**  
1. Pull clean versions of poky/master and meta-raspberrypi/master
2. In "local.conf" add:
- `MACHINE ?= "raspberrypi4-64"`
- `ENABLE_UART="1"`
- `ENABLE_DWC2_HOST = "1"`
- `IMAGE_FSTYPES="rpi-sdimg"`
- `CORE_IMAGE_EXTRA_INSTALL+="openssh openssh-sshd openssh-sftp openssh-sftp-server"`
3. `bitbake core-image-minimal`
6. Use `win32diskimager` to flash an SD card with the resulting image
7. Install the SD card in the RPi4
8. Boot the board
9. Log in via SSH as root
10. Plug in my backlit USB keyboard
11. Connect a 3.3v TTL to USB converter to the UART on the GPIO header 
12. Attempt connection to the UART ttl  
  
Upon discovery of the issue, I mounted the `boot` partition in windows to verify the `config.txt` file and added to the end of the file were the lines `enable_uart=1` and `dtoverlay=dwc2,dr_mode=host`. I also mounted the partition to the boot directory on the resulting image and verified the same results.
  
**Describe the results you received:**  
The USB did not light on the keyboard on the CM4. Additionally, the ttl UART did not function with either board.  
  
**Describe the results you expected:**  
The USB to be powered and operational on the CM4 and the ttl UART to be operation on both the CM4 and the RPi4.
  
**Additional information you deem important (e.g. issue happens only occasionally):**  
There are more things in the "config.txt" I could probably try, but thus far, anything I've tried to modify in config.txt has not applied.

ghuband, 2021-03-10:
> @agherzan I can confirm that this works on a 32bit configuration. I tested on a CM4 with 4Gb of RAM without eMMC, with WiFi.

How did you build the 32-bit image for the cm4?

aurelian17, 2021-03-10:
@ghuband, just use MACHINE ?= "raspberrypi4" I don't need 64 bit for the 4Gb RAM CM4, so I used the 32 bit image. Please give it a try and tell us if solves your problem.

ghuband, 2021-03-10:
Was able to get usb working with 64-bit core-image-base.

aurelian17, 2021-03-10:
By the way, did you enable ENABLE_DWC2_HOST = "1" ?
Even if this sounds strange, the USB HOST is disabled by default on CM4.

asuhler, 2021-03-10:
@aurelian17 
> By the way, did you enable ENABLE_DWC2_HOST = "1" ?
> Even if this sounds strange, the USB HOST is disabled by default on CM4.

Turns out something wasn't being pulled in in the `core-image-minimal` for USB, I would assume the USB driver based on the above debug. I can confirm @ghuband 's result that `core-image-base` worked as expected.

aurelian17, 2021-03-10:
Yes, indeed my image is derived from core-image-base. In order to do that just add include recipes-core/images/core-image-base.bb in your image recipe.

asuhler, 2021-03-10:
This solve many hours of headache, thanks all! If only all the common guides didn't use "core-image-minimal" for their base builds.

ghuband, 2021-03-10:
Thanks, we will use core-image-base.

closed 2021-03-10 by ghuband

jonathanhaigh, 2021-04-07:
FWIW, I had the same issue with USB2.0 on the CM4 using core-image-minimal and solved it by just adding the `kernel-module-dwc2` and `kernel-module-roles` packages.

nmaas87, 2021-04-08:
Thanks @jonathanhaigh - thats exactly what I am needing right now - trying to get USB working in CM4 using core-image-minimal.
I tried to just add `kernel-module-dwc2` and `kernel-module-roles` to the `IMAGE_INSTALL_append` - but that did just fail. Then I tried to create a custom image from core-image-minimal. In this package list, it can find (and add)  `kernel-module-dwc2`  - however `kernel-module-roles` is not available. Am I doing something wrong here? :) Thanks!

nmaas87, 2021-04-08:
It worked already. Just add a IMAGE_INSTALL_append = 'kernel-module-dwc2 kernel' (the generic kernel package seems to be important and I guess it would add it automatically if left out) and create a new image, worked. Enabling the DWC2 as described here: https://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#enable-usb-host-support

RexLBrownSr, 2022-03-22:
@nmaas87 -- Thanks! Wow after searching everywhere, trying this worked!