agherzan/meta-raspberrypi #826: Proper way of using external overlays

nandra, 2021-03-10:
I would like to ask about possibility to include to build custom overlays. To my experience there exists only one way which is to patch linux-raspberrypi and add custom overlay here.

I was looking at approach to have separate recipe which inherit devicetree class which can build dtbo. Then I've added to local.conf:
```
RPI_KERNEL_DEVICETREE_OVERLAYS_append = " \
      mybest-pi-overlay.dtbo \
```

Problem is then kernel fails due to the fact that such overlay doesn't exists. Is there other way possible to have it done like described? Thanks.

MikeLooijmans, 2021-03-10:
You can compile your own dts files outside the kernel, just inherit devicetree in a recipe and provide the dts in SRC_URI

To install these on the filesystem, add them in your image recipe (not local.conf or so) like so:
RPI_KERNEL_DEVICETREE_OVERLAYS += "overlays/my-overlay.dtbo"

That's how I do it. Also makes it take only a few seconds to copmile and deploy an updated overlay.

MikeLooijmans, 2021-03-11:
RPI_KERNEL_DEVICETREE_OVERLAYS

Must be in your *image* recipe. It has no effect in other places.

nandra, 2021-03-11:
OK thanks that does the trick. :+1:

closed 2021-03-11 by nandra

agherzan, 2021-03-11:
Let me explain this a bit. RPI_KERNEL_DEVICETREE_OVERLAYS is defined at the machine level and taken into consideration by two recipes - kernel and image. Kernel for the compilation process (as it assumes that the dts are there) and image for the deployment. The above would work because it especially only adds a custom blob to the deployment step. But that would assume you are using the sdcard class. For wic would most probably not work. I think for custom device trees we would need to make this a bit clearer - but for now the above would work.

MikeLooijmans, 2021-03-11:
I'm sure it works for wic, since it's the only thing I'm using at the moment.

agherzan, 2021-03-11:
Right. It does work. Just realised. Nevertheless, this needs some improvement.

MikeLooijmans, 2021-03-11:
On other systems, the devicetree recipes just install into the /boot/devicetree location, and simply adding a devicetree package to the image in some way (directly or through some package-group or machine dependency) makes it appear in the root filesystem. So it doesn't even need the "deploy" step.
Problem with the Pi is that the devicetrees (and kernel) are on another partition.

A simple addition would be to just add all dtbo files in the $DEPLOYDIR/devicetree directory to the overlays directory on the boot partition. That'd remove the RPI_KERNEL_DEVICETREE_OVERLAYS out of the equation, and one can simply add devicetree overlays by just adding a dependency.

agherzan, 2021-03-16:
That would work yes but I find it a bit hacky. I'll think about it. I feel that we should clean this a bit more.

MikeLooijmans, 2021-03-17:
The issue with DT overlays is that they're on "application" level instead of machine level.

Another approach is to ignore the RPi firmware DT handling and have the kernel load the custom devicetree. That has the advantage that your overlays reside on the rootfs and can simply be injected from userspace. And that works for all boards and isn't RPi specific.