agherzan/meta-raspberrypi #831: CM4 Unable to configure additional UARTs via config.txt

asuhler, 2021-03-17:
**Description**
I have been trying to configure additional UARTs (UART2,3,4,5) in my `core-image-base` through the `config.txt` to no avail. The configuration is in the `config.txt` just fine, but the devices do not show up in `/dev/ttyAMAx`. I do not know enough to dig into the device trees myself to investigate, unfortunately, since I'm fairly new. 

**Steps to reproduce the issue:**
1. Pull clean versions of poky/master and meta-raspberrypi/master (plus qt5, openembedded-core, other dependencies)
2. `source oe-init-build-env`
3. In "local.conf" add:
- `MACHINE ?= "raspberrypi4-64"`
- `ENABLE_UART="1"`
- `ENABLE_DWC2_HOST = "1"`
- `IMAGE_FSTYPES="rpi-sdimg"`
4. Add the required layers to `bblayer.conf`
5. `bitbake core-image-base` (THIS was my issue in the past with config.txt not applying, someone kindly pointed out that `core-image-minimal` would not work to apply config.txt`)
6. Set the j2 jumper for disable eMMC boot and plug in the micro-usb cable to my windows desktop
7. Connect the CM4 board to windows with the `rpiboot` utility
8. Use `win32diskimager` to flash the eMMC with my image
9. Edit `config.txt` in the `boot` partition to add the following config:
-  dtoverlay=dwc2,dr_mode=host
- `enable_uart=1`
- `dtoverlay=uart2`
- `dtoverlay=uart3`
- `dtoverlay=uart4`
- `dtoverlay=uart5`
10. Power off the board, remove the jumper from J2 and unplug the micro-usb cable
11. Boot the board normally
12. run `ls /dev/ttyAMA*`

Raspberrypi OS:
For Raspberrypi OS testing, I installed Raspberrypi OS Lite (the most recent version, kernel 5.10) on the CM4 and added the same configuration to `config.txt`.

**Describe the results you received:**
I expected to see `/dev/ttyAMA0 /dev/ttyAMA2 /dev/ttyAMA3 /dev/ttyAMA4 /dev/ttyAMA5` in the `/dev/` directory. This would indicate that the UARTs have been enabled in the device tree. 

**Describe the results you expected:**
In Raspberrypi OS , the configuration took just fine and I successfully enabled all UARTs (`/dev/ttyAMA0-5`, excluding 1).

In Yocto, only the default UART (`/dev/ttyAMA0`) was enabled. I verified the configuration was in the `config.txt` by mounting `mmcblk0p1` on the yocto build and checking the config. 

**Additional details (revisions used, host distro, etc.):**
I am doing all of this on a CM4 with 2Gb ram, 16Gb eMMC, and no wifi/bluetooth. I will also note that the `dtoverlay=dwc2,dr_mode=host` applied fine to enable the USB ports. 

I will also note I attempted to use `dtparam=uart1=on` to no avail on either Raspberrypi OS or Yocto. 

Any help would be appreciated, thanks!

asuhler, 2021-03-17:
Update: I verified that the issue exists in both 64 bit and 32 bit versions of the `machine=raspberrypi4` build. I'm sure there's an issue with my process more than a bug, but I cannot figure out what is not correct.

kraj, 2021-03-17:
can you check and compare what overlays are there in both yocto and raspbian in `/boot/overlays/`

asuhler, 2021-03-18:
@kraj Thanks for the reply! I've USED embedded linux for a few years, this is my first time looking harder at building embedded linux, so I knew I was missing something. I checked the overlays directory in the CM4 and found this:
[image]
I do not see any UARTs listed in the overlays, whereas with the Raspberrypi OS, I get over a full 100 line terminal of overlays and right at the bottom are the UARTs I was looking for:
[image]

I will work on getting `dtoverlay` results, it looks like `dtc` and `dtc-misc` did not pull in `dtoverlay`. I will start searching, just wanted to give this update!

asuhler, 2021-03-18:
I can confirm this is the issue, copying over the `uartx.dtbo` files from the raspberrypi OS to the `/boot/overlays` of the yocto build solved the issue. I now have the expected `/dev/ttyAMAx` devices in yocto after adding the `dtoverlay=uartx` to `config.txt`. 
[image]

kraj, 2021-03-18:
right, then we need to build these device tree overlays in yocto when building kernel. add them to 

`RPI_KERNEL_DEVICETREE_OVERLAYS` in `conf/machine/include/rpi-base.inc`

asuhler, 2021-03-18:
@kraj This is what I was looking for, thanks! Seeing as this was more a "yocto knowledge" issue than a meta-raspberrypi issue, is there a better place I can ask these questions in the future instead of opening issues? Thanks!

closed 2021-03-18 by asuhler

kraj, 2021-03-18:
@asuhler this was perfect question for meta-raspberrypi so no worries. Thanks for asking

agherzan, 2021-03-23:
@asuhler are you able to follow up with a PR to include the specific overlays?

asuhler, 2021-03-23:
@agherzan Unless we want to include the overlays in the `core-image-base` target, I don't think it's needed. I was able to add the `uartx.dtb0` files in the build using:  
`RPI_KERNEL_DEVICETREE_OVERLAYS_append = " overlays/uart1.dtbo overlays/uart2.dtbo overlays/uart3.dtbo overlays/uart4.dtbo overlays/uart5.dtbo"`  
  
I will note that using `RPI_KERNEL_DEVICETREE_OVERLAYS += overlays/uart1.dtbo` resulted in ONLY `uart1.stbo` being installed in `/boot/overlays`, but switching to `_append` fixed that issue.