agherzan/meta-raspberrypi #837: What "other data" is located in the 4MiB between 0 and IMAGE_ROOTFS_ALIGNMENT?

evading, 2021-03-28:
In `classes/sdcard_image-rpi.bbclass` there is a comment:
```
#    0                      -> IMAGE_ROOTFS_ALIGNMENT         - reserved for other data
```
I'm curious about what "other data" that is as I would like to use that space for two redundant u-boot environments.
I realise there is an MBR in there in the beginning but that will not consume the full 4MiB, right?

Is it safe to assume that the space 3MiB to 4MiB is available for this?

Thanks for an awesome layer!

Best regards,
Einar

agherzan, 2021-04-03:
That was only done for performance reasons on SD cards. And indeed, you should be able to use the remaining free space. What is your usecase?

closed 2021-04-03 by agherzan

evading, 2021-04-06:
We have found that placing the u-boot environment on a FAT partition is a bit fragile. And, with the partition layout we're using, it forces us to mount the boot partition R/W. Also I don't think u-boot supports redundant envs in FAT.

To get around that we plan to put two u-boot environments straight on the device at some offset in the first 4MiB, like at 3MiB and 3.5MiB.