agherzan/meta-raspberrypi #839: Raspberty pi 4 not booting

Kottoz, 2021-04-04:
i follow steps to bitbake the core image base , using `bitbake core-image-base` everything is built successfully, then i use `sudo bmaptool copy core-image-base-raspberrypi4-64.wic.bz2 /dev/sdb' to my sd card, after that raspberrypi not booting.

this is my `local.conf`
```
#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
#
#MACHINE ?= "beaglebone-yocto"
#MACHINE ?= "genericx86"
#MACHINE ?= "genericx86-64"
#MACHINE ?= "edgerouter"
#
# This sets the default machine to be qemux86-64 if no other machine is selected:
MACHINE ??= "raspberrypi4-64"
#IMAGE_FSTYPES="tar.xz ext3 rpi-sdimg wic wic.bmap"
#IMAGE_INSTALL_append = " linux-firmware-bcm43455 wpa-supplicant"
#
# Where to place downloads
#
# During a first build the system will download many different source code tarballs
# from various upstream projects. This can take a while, particularly if your network
# connection is slow. These are all stored in DL_DIR. When wiping and rebuilding you
# can preserve this directory to speed up this part of subsequent builds. This directory
# is safe to share between multiple builds on the same machine too.
#
# The default is a downloads directory under TOPDIR which is the build directory.
#
#DL_DIR ?= "${TOPDIR}/downloads"
DL_DIR ?= "/home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/downloads"

#
# Where to place shared-state files
#
# BitBake has the capability to accelerate builds based on previously built output.
# This is done using "shared state" files which can be thought of as cache objects
# and this option determines where those files are placed.
#
# You can wipe out TMPDIR leaving this directory intact and the build would regenerate
# from these files if no changes were made to the configuration. If changes were made
# to the configuration, only shared state files where the state was still valid would
# be used (done using checksums).
#
# The default is a sstate-cache directory under TOPDIR.
#
#SSTATE_DIR ?= "${TOPDIR}/sstate-cache"
SSTATE_DIR ?= "/home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/sstate-cache"
#
# Where to place the build output
#
# This option specifies where the bulk of the building work should be done and
# where BitBake should place its temporary files and output. Keep in mind that
# this includes the extraction and compilation of many applications and the toolchain
# which can use Gigabytes of hard disk space.
#
# The default is a tmp directory under TOPDIR.
#
[... 146 lines cut ...]

here is my 'bblayers.conf'

```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky/meta \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky/meta-poky \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky/meta-yocto-bsp \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/layers/meta-raspberrypi \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/layers/meta-openembedded/meta-oe \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/layers/meta-openembedded/meta-multimedia \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/layers/meta-openembedded/meta-networking \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/layers/meta-openembedded/meta-python \
  "

```

i have tried to download rasbian on sd card and everything is good, it is not a hw problem, so how can i solve that.

Kottoz, 2021-04-04:
Edit i try the last steps, with new HW device, raspberry pi 3, and it works fine. and boots properly, so what is  wrong  with RPI4?

kraj, 2021-04-04:
I just booted today from master branches of layer and it works all fine for rpi4 ( I use yoe distro )

Kottoz, 2021-04-04:
I will test changing to master

Kottoz, 2021-04-06:
@kraj i have tested master poky. 
here is my `local.conf`
```
MACHINE = "raspberrypi4-64"

RPI_USE_U_BOOT = "1"
LICENSE_FLAGS_WHITELIST = "commercial"
IMAGE_INSTALL_append = " kernel-image kernel-modules"
INHERIT += "image-buildinfo"
IMAGE_ROOTFS_EXTRA_SPACE = "327680"

DISTRO_FEATURES_append = " systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED += "sysvinit"
VIRTUAL-RUNTIME_init_manager = "systemd"
VIRTUAL-RUNTIME_initscripts = "systemd-compat-units"
DL_DIR ?= "/home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/downloads"
SSTATE_DIR ?= "/home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/sstate-cache"
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_ipk"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"
PACKAGECONFIG_append_pn-qemu-system-native = " sdl"
CONF_VERSION = "1"
```

here is my `bblayers.conf`

```
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/poky/meta \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/poky/meta-poky \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/poky/meta-yocto-bsp \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/layers/meta-raspberrypi \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/layers/meta-openembedded/meta-oe \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/layers/meta-openembedded/meta-python \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/layers/meta-openembedded/meta-networking\
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/layers/meta-openembedded/meta-perl \
  /home/kotoz/Desktop/Embedded_linux/yocto_for_rpi/poky_master/layers/meta-openembedded/meta-multimedia \
  "
```
images was built successfully with no errors.

i use the following command to boot from sd card.
`sudo bmaptool copy ./tmp/deploy/images/raspberrypi4-64/core-image-base-raspberrypi4-64.wic.bz2 /dev/sdb`

when i pluged the sd card then the power cable the green flash was blinking, then make a cracking flashes then off. 
i use hdmi screen, i goes into power save mode. 

i have used the last steps with raspberry pi 3 and it works properly. so what is the problem ?

kraj, 2021-04-06:
Can you try after removing below line from your `local.conf`

```
RPI_USE_U_BOOT = "1"
```

Kottoz, 2021-04-06:
@kraj 
i have already test that,  just keep the default `local.conf`, and edit `MACHINE = "raspberrypi4"` . 
the first trial my pi has connected and detected the hdmi, rainbow background start to appears for 2 seconds then nothing to show. 
i remove the power then plug it again, nothing happens and not detects hdmi. 

my pi is raspberrypi4 , 2G rams, using 32 G class 10 sd card.

Kottoz, 2021-04-06:
i will make another test keep `local.conf` is default but update the following 
```
MACHINE = "raspberrypi4"

#RPI_USE_U_BOOT = "1"
LICENSE_FLAGS_WHITELIST = "commercial"
IMAGE_INSTALL_append = " kernel-image kernel-modules"
INHERIT += "image-buildinfo"
IMAGE_ROOTFS_EXTRA_SPACE = "327680"

```

Kottoz, 2021-04-06:
@kraj 
i have used a shared `download` and `sstate-cache` folders between different pokies branches and also different hardware.
should i isolate each `download` and `sstate-cache` for each poky. 

what i mean that i have poky with dunfell branch and another in different directory with master. 
is this the problem ?

kraj, 2021-04-06:
> @kraj
> i have used a shared `download` and `sstate-cache` folders between different pokies branches and also different hardware.
> should i isolate each `download` and `sstate-cache` for each poky.

I think its fine to share these directories.

> 
> what i mean that i have poky with dunfell branch and another in different directory with master.
> is this the problem ?
> 
> i will make a try to isolate both by commenting `DL` and `SSTATE-CACHE` inside `local.conf`
> 
> this works ! but how

What works ? I did not understand

Kottoz, 2021-04-06:
I was using shared sstate-cache between all my projects, which are based on
poky dunfell branch. I have created another project based on poky master,
and used the shared sstate-cache and downloads.
After i have removed the shared sstate-cache and downloads, just commented
them.
The device is booted properly.

But after remove the power and plug it again it doesn't work hdmi monitor
is in power saving mode. I have also noticed that the device has taken an
ip address as i connect the pi to router via ethernet cable.

I suspected that i have a problem in my mini  hdmi socets then i will try
to use another one and post my feedback

agherzan, 2021-04-06:
Note for future: prefer attaching configuration files as opposed to pasting long ones in the issue thread.

On topic though, are you sure that initially, the board was not booting at all? Maybe it just behaved as in the case you described above - for which you suspect the HDMI cable/port.

kraj, 2021-04-06:
> Note for future: prefer attaching configuration files as opposed to pasting long ones in the issue thread.
> 
> On topic though, are you sure that initially, the board was not booting at all? Maybe it just behaved as in the case you described above - for which you suspect the HDMI cable/port.

If you look at my suggestion I asked to disable u-boot and fall back to direct boot loader instead 
It might be worth testing if
This is the reason I have had enough troubles with  unit on master lately that I have disabled it in yoe distro for now

Kottoz, 2021-04-06:
i have updated the issue. 
every thing is worked well. 
thanks 
i will close this issue

closed 2021-04-06 by Kottoz