agherzan/meta-raspberrypi #840: vc4graphics not working on model 3B raspberrypi3-64

wbochra, 2021-04-06:
**Description**
I am building an core-image-sato with MACHINE ??= "raspberrypi3-64" 
I was toled the vc4graphics  is enabled by default no and it is but I didn't see any graphics when I boot the board with putty (ethernet connection)
so I tried to add this variabel to the local.conf DISABLE_VC4GRAPHICS = "1" now I have this error 

ERROR: Nothing RPROVIDES 'libgl' (but /home/ubuntu/poky/meta/recipes-gnome/gtk+/gtk+3_3.24.25.bb RDEPENDS on or otherwise requires it)
mesa RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgbm set to mesa-gl, not mesa
mesa-gl RPROVIDES libgl but was skipped: PREFERRED_PROVIDER_virtual/libgl set to mesa, not mesa-gl
NOTE: Runtime target 'libgl' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libgl']
NOTE: Runtime target 'libnss-mdns' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['libnss-mdns', 'avahi', 'gtk+3', 'libgl']
NOTE: Runtime target 'packagegroup-base-extended' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']
ERROR: Required build target 'core-image-sato' has no buildable providers.
Missing or unbuildable dependency chain was: ['core-image-sato', 'packagegroup-base-extended', 'libnss-mdns', 'avahi', 'gtk+3', 'libgl']

wbochra, 2021-04-07:
I moved this line include conf/machine/include/rpi-default-providers.inc from rpi-base.inc to raspberrypi3-64.conf now I can make it to the build with this configurations
`Loading cache: 100% |############################################| Time: 0:00:00
Loaded 3962 entries from dependency cache.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION = "1.49.2"
BUILD_SYS = "x86_64-linux"
NATIVELSBSTRING = "universal"
TARGET_SYS = "aarch64-poky-linux"
MACHINE = "raspberrypi3-64"
DISTRO = "poky"
DISTRO_VERSION = "3.2+snapshot-ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
TUNE_FEATURES = "aarch64 armv8a crc cortexa53"
TARGET_FPU = ""
meta
meta-poky
meta-yocto-bsp = "master:ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
meta-raspberrypi = "master:d1f191ed3018e0b130311e443f014e30d2f5ed97"
meta-oe
meta-python
meta-multimedia
meta-networking
meta-perl
meta-gnome
meta-xfce = "master:ba2576ed79a09879f714ccf7f4fa32259c18ab8c"
meta-test = "master:ea455ca8671d3bc2a1097989bfaabe92f3ca37ab"
`
the error now:
ERROR: mesa-gl-2_21.0.0-r0 do_configure: meson failed
ERROR: mesa-gl-2_21.0.0-r0 do_configure: Execution of '/home/ubuntu/build/tmp/work/cortexa53-poky-linux/mesa-gl/2_21.0.0-r0/temp/run.do_configure.20999' failed with exit code 1:

ERROR: Neither directory contains a build file meson.build.
WARNING: exit code 1 from a shell command.

ERROR: Logfile of failure stored in: /home/ubuntu/build/tmp/work/cortexa53-poky-linux/mesa-gl/2_21.0.0-r0/temp/log.do_configure.20999

nicolasb565, 2021-05-13:
Hi wbochra, I had a similar issue, the reason was that the kernel modules we're not being built/added to the image.

Just doing that fixed it for me.
IMAGE_INSTALL_append +=    " kernel-modules "

wbochra, 2021-05-14:
@nicolasb565 Hi I gaved up on the idea a while ago. thank you anyway. can you please help me in somthing else had tried to boot your raspberry pi image (yocto build) on qemu ?

nicolasb565, 2021-05-14:
No never tried that unfortunately.

agherzan, 2021-11-14:
I've tested vc4 support in master on rpi3 and rpi4. Support seems fine, Please retest and if the issue persists please create another issue where you describe exactly the revisions, the setup and configurations you have used.

closed 2021-11-14 by agherzan