agherzan/meta-raspberrypi #841: Raspberrypi4 WiFi is not working using core-image-full-cmdline

Kottoz, 2021-04-07:
As i see in the documentation that wifi is already added by default to `core-image-base`, i have test it and wifi works well. However when i use `core-image-full-cmdline`, wifi doesn't work.
when i run `ifup wlan0`
```
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
Could not read interface wlan0 flags: No such device
WEXT: Could not set interface 'wlan0' UP
wlan0: Failed to initialize driver interface
```
run `ifconfig -a`
```
eth0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether dc:a6:32:3c:fd:0e  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 2  bytes 140 (140.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2  bytes 140 (140.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
i have checked #576, #703, and #729. 
so, could i add `IMAGE_INSTALL_append = " linux-firmware-bcm43455` 
[bootlog.txt](https://github.com/agherzan/meta-raspberrypi/files/6268589/bootlog.txt)
[local.txt](https://github.com/agherzan/meta-raspberrypi/files/6268592/local.txt)
[bblayers.txt](https://github.com/agherzan/meta-raspberrypi/files/6268593/bblayers.txt)

Kottoz, 2021-04-07:
After some research i find that 

> MACHINE_EXTRA_RRECOMMENDS variable affects only images based on packagegroup-base, which does not include the core-image-minimal or core-image-full-cmdline images. 
So i have tried to edit `raspberrypi4.conf` file by replacing `MACHINE_ESSENTIAL_EXTRA_RDEPENDS` instead of `MACHINE_EXTRA_RRECOMMENDS`, but it doesn't work

Kottoz, 2021-04-08:
@kraj, could you give me hints!

kraj, 2021-04-08:
> @kraj, could you give me hints!

Use yoe distro and see if it works ok with `yoe-simple-image`

agherzan, 2021-04-12:
Does it work with `core-image-base`?

Kottoz, 2021-04-13:
@agherzan  ya, it works fine.

closed 2021-04-13 by agherzan

gitgonewithwind, 2021-08-30:
@Kottoz 
do you find the solution? I have the some issue with core-image-full-cmdline ?
I do not know how to enable the wifi work.
I seems that you closed the issue, but I read  them , I do not find your solution.
before I do not use the pi 4 wiki function with core-image-full-cmdline, but one of our projects will use it
any help?
I have to switch to core-image-base?
any help?

agherzan, 2021-09-03:
I've replied in the issue you've created above.

gitgonewithwind, 2021-09-04:
@agherzan thanks