agherzan/meta-raspberrypi #842: libva build fails when using userland

matt-hammond-001, 2021-04-08:
**Description**

I'm building an image for a raspberrypi4 based on **dunfell** and the `core-image-minimal` recipe. Including `libva` in the recipe fails if I disable vc4graphics, despite including `userland`. It appears to be the problem first described in #391.

**Steps to reproduce the issue:**
1. Start with poky on dunfell branch
2. Add `MACHINE_FEATURES_remove="vc4graphics"` to local.conf
3. Add `userland` and `libva` to `IMAGE_INSTALL`

**Describe the results you received:**

Complains that  `/usr/include/KHR/khrplatform.h` is missing, when trying to build image, e.g.
```
usr/include/GL/glext.h:56:10: fatal error: KHR/khrplatform.h: No such file or directory
|    56 | #include <KHR/khrplatform.h>
```

**Apparent fix**

I've been able to fix it locally by creating `libva_2.6.1.bbappend` within my own layer and putting in it:
```
DEPENDS_append_rpi = " ${@bb.utils.contains('MACHINE_FEATURES', 'vc4graphics', '', 'userland', d)}"
```
... basically the same solution as here: https://github.com/agherzan/meta-raspberrypi/commit/ac16b0e9d713bcc4ee0e976259064d1a318a5aac

Assuming this is the correct approach to fix, my guess is that it is better if the fix is done here for dunfell and other branches.

agherzan, 2021-04-12:
Yes. Could you create an MR that cherry-picks the specific fix on `dunfell` and raises it against the `dunfell` branch?

matt-hammond-001, 2021-06-08:
Done ... I think?

matt-hammond-001, 2021-06-08:
Now you've merged, I've deleted the branch. Many thanks

PR #848 (2021-06-08): libva: Fix for when using `userland`
libva: Fix for when using `userland`

Fixes #842 - failing build of libva when using `userland` recipe.
Fix is same as: https://github.com/agherzan/meta-raspberrypi/commit/ac16b0e9d713bcc4ee0e976259064d1a318a5aac
closed 2021-06-08 by agherzan