agherzan/meta-raspberrypi #846: Raspberry Pi CM4 not booting with U-Boot

havok13888, 2021-04-14:
[image]

Image with RPI_USE_U_BOOT = "0" build and boots successfully on the PI4B and CM4. 
Image with RPI_USE_U_BOOT = "1" build and boots successfully on the PI4B. It does not boot with a CM4. 

Using default boot script with latest gatesgarth.

[image]

thomas-at-rock7, 2021-04-16:
I think the issue is related to u-boot not having a supported identifier for the Raspberry CM4 module yet. So the wrong device tree blob (DTB) is loaded.

At the u-boot prompt you can issue `env print fdtfile` and it will print a loaded DTB.
```
U-Boot> env print fdtfile
fdtfile=bcm283x-rpi-other.dtb
```
The DTB should be `bcm2711-rpi-cm4.dtb`.

I believe it was added to the u-boot source [here](https://source.denx.de/u-boot/u-boot/-/commit/561bce3522bd1df5f6b10fbae5a6ea5c0b131ff9) and it was included in version 2021.04.

Unfortunately the Poky master branch only supports version 2021.01 at the moment. So you may have to wait or do your own recipe to bring in 2021.04 or apply the appropriate patches.

closed 2021-04-20 by havok13888

agherzan, 2021-04-23:
Thanks, @havok13888 for the update.