agherzan/meta-raspberrypi #847: Do FATPAYLOAD and DEPLOYPAYLOAD work on WIC images that uses bootimg-partition instead of sdcard_image class?

timemaster5, 2021-04-14:
**Description**

I am trying to deploy some files to a /boot partition. 
According to documentation, FATPAYLOAD or DEPLOYPAYLOAD should be enough to do that. 

Th proble is that I don't use .sdcard format, but instead I use .wic images.

Do you have any workaround on how to extend FATPAYLOAD and DEPLOYPAYLOAD to work even with .wic images, please?

**Steps to reproduce the issue:**
set DEPLOYPAYLOAD to whatewertestfile (must be located in IMAGE_DEPLOY_DIR)
set IMAGE_FSTYPES to wic

**Describe the results you received:**
working image without any extra content deployed to /boot partition

**Describe the results you expected:**
working image with extra content described in DEPLOYPAYLOAD directive

**Additional information you deem important (e.g. issue happens only occasionally):**
wic images uses bootimg-partition source plugin which is not extended by sdcard_image-rpi.bbclass

timemaster5, 2021-04-14:
Ok, I have managed to resolve this using https://www.yoctoproject.org/docs/2.1/ref-manual/ref-manual.html#var-IMAGE_BOOT_FILES, which worked great. So maybe only some note in the documentation is needed? Or I am blind.

Closing anyway, just for the reference.

closed 2021-04-14 by timemaster5

agherzan, 2021-04-23:
That is indeed the case. `FATPAYLOAD` is only for the sdcard bbclass. `wic` comes with the mechanisms defined in Yocto - IMAGE_BOOT_FILES.