agherzan/meta-raspberrypi #851: SSH not working in Rpi0 and Rpi0-w hardware with both yocto dunfell and gatesgarth branch

arghyaBiswas05, 2021-04-27:
**SSH not working in Rpi0 and Rpi0-w hardware with both yocto dunfell and gatesgarth branch**

I tried to compiled for both dunfell and gatesgarth branch for MACHINE = "raspberrypi0-wifi" and MACHINE = "raspberrypi0". All of the other features are working perfectly except SSH. From my PC I am not able to login into the Rpi via SSH. But if I try to log in to my PC from Rpi via SSH then it is working fine. So as per my knowledge, SSH-server is not working properly.

**Steps to reproduce the issue:**
1. build yocto project for dunfell or gatesgarth brnach
2. use the machine as raspberrypi0 or raspberrypi0-wifi
3. boot the os and try to do SSH login from another computer to the Raspberry Pi

**Additional details (revisions used, host distro, etc.):**
in Local.conf I used 

`EXTRA_IMAGE_FEATURES = "debug-tweaks tools-sdk tools-debug"`
`IMAGE_INSTALL_append = " i2c-tools openssh openssh-sshd openssh-sftp openssh-sftp-server"`
`IMAGE_INSTALL_append = " iw wpa-supplicant packagegroup-base module-init-tools"`
`IMAGE_INSTALL_append = " linux-firmware-bcm43430 bluez5 bridge-utils hostapd iptables"`

`ENABLE_I2C = "1"`

`KERNEL_MODULE_AUTOLOAD_rpi += "i2c-dev i2c-bcm2708"`

> ** Able to login into other computer from Raspberry pi via SSH, but not able to login into Raspberry Pi from other computer.

Please help to debug the issue

kraj, 2021-04-27:
add `EXTRA_IMAGE_FEATURES += "ssh-server-dropbear ` to your `local.conf`

agherzan, 2021-04-27:
He uses openssh-sshd. Have you tries to check the ssh logs? Can you ping the machine?

kraj, 2021-04-27:
> He uses `openssh-sshd`. @arghyaBiswas05 Have you tried to check the ssh logs? Can you ping the machine?

Yes, that's why I am asking to use dropbear and see if that works or not. ( provided it has network up and IP assigned )

arghyaBiswas05, 2021-04-27:
My network was up and also assigned an IP perfectly. I tried to ping the machine and it was responding fine...

FYI I am able to do SSH login from Rpi to my PC, but not able to login from PC to Rpi.

** let me try once with dropbear and check.

arghyaBiswas05, 2021-04-30:
Thanks a lot, @kraj . After use, the ssh-dropbear everything is working fine.. :)

closed 2021-04-30 by arghyaBiswas05