agherzan/meta-raspberrypi #853: Pi4 u-boot USB controller not found

havok13888, 2021-04-29:
Hi, whenever the system boots and the usb start command is run in u-boot it is unable to find the usb controller. I can see the usb devices running and mounted when I'm in the OS

[image]

Using:
- u-boot 2021.04
- all layers on gatesgarth
- Building for pi4 and cm4

I have the following enabled in u-boot def config. 

CONFIG_CMD_USB=y
CONFIG_USB=y
CONFIG_DM_USB=y
CONFIG_DM_USB_GADGET=y
CONFIG_USB_XHCI_HCD=y
CONFIG_USB_XHCI_PCI=y
CONFIG_USB_KEYBOARD=y
CONFIG_USB_GADGET=y
CONFIG_USB_GADGET_MANUFACTURER="FSL"
CONFIG_USB_GADGET_VENDOR_NUM=0x0525
CONFIG_USB_GADGET_PRODUCT_NUM=0xa4a5
CONFIG_USB_GADGET_DWC2_OTG=y
CONFIG_USB_GADGET_DOWNLOAD=y

Thanks

havok13888, 2021-05-05:
The ultimate goal was to run this on a CM4, there seems to be an issue there too but a different one. I have an issue opened at linux-raspberrypi: https://github.com/raspberrypi/linux/issues/4325

agherzan, 2021-06-08:
@havok13888 As far as I remember, uboot didn't have USB support (maybe that changed, and all we need is a backport or a version bump). I encourage you to follow up on the uboot mailing list for this issue.

closed 2021-06-08 by agherzan