agherzan/meta-raspberrypi #855: Raspberry Pi HD quality camera support

sss22213, 2021-05-13:
hi, I create new diff for Raspberry Pi HD quality camera.

```bash=
diff --git a/conf/machine/include/rpi-base.inc b/conf/machine/include/rpi-base.inc
index 77cada7..a800078 100644
--- a/conf/machine/include/rpi-base.inc
+++ b/conf/machine/include/rpi-base.inc
@@ -31,6 +31,7 @@ RPI_KERNEL_DEVICETREE_OVERLAYS ?= " \
     overlays/justboom-digi.dtbo \
     overlays/i2c-rtc.dtbo \
     overlays/imx219.dtbo \
+    overlays/imx477.dtbo \
     overlays/iqaudio-dac.dtbo \
     overlays/iqaudio-dacplus.dtbo \
     overlays/mcp2515-can0.dtbo \
diff --git a/recipes-bsp/bootfiles/rpi-cmdline.bb b/recipes-bsp/bootfiles/rpi-cmdline.bb
index 40a9949..3ebd1e6 100644
--- a/recipes-bsp/bootfiles/rpi-cmdline.bb
+++ b/recipes-bsp/bootfiles/rpi-cmdline.bb
@@ -13,6 +13,8 @@ CMDLINE_SERIAL ?= "${@oe.utils.conditional("ENABLE_UART", "1", "console=serial0,
 
 CMDLINE_CMA ?= "${@oe.utils.conditional("RASPBERRYPI_CAMERA_V2", "1", "cma=64M", "", d)}"
 
+CMDLINE_CMA ?= "${@oe.utils.conditional("RASPBERRYPI_HD_CAMERA", "1", "cma=64M", "", d)}"
+
 CMDLINE_PITFT ?= "${@bb.utils.contains("MACHINE_FEATURES", "pitft", "fbcon=map:10 fbcon=font:VGA8x8", "", d)}"
 
 # Add the kernel debugger over console kernel command line option if enabled
diff --git a/recipes-bsp/bootfiles/rpi-config_git.bb b/recipes-bsp/bootfiles/rpi-config_git.bb
index de43c4b..b537bfb 100644
--- a/recipes-bsp/bootfiles/rpi-config_git.bb
+++ b/recipes-bsp/bootfiles/rpi-config_git.bb
@@ -184,10 +184,16 @@ do_deploy() {
 
     # Choose Camera Sensor to be used, default imx219 sensor
     if [ "${RASPBERRYPI_CAMERA_V2}" = "1" ]; then
-        echo "# Enable Sony RaspberryPi Camera" >> $CONFIG
+        echo "# Enable Sony RaspberryPi Camera(imx219)" >> $CONFIG
         echo "dtoverlay=imx219" >> $CONFIG
     fi
 
+    # Choose Camera Sensor to be used, default imx477 sensor
+    if [ "${RASPBERRYPI_HD_CAMERA}" = "1" ]; then
+        echo "# Enable Sony RaspberryPi Camera(imx477)" >> $CONFIG
+        echo "dtoverlay=imx477" >> $CONFIG
+    fi
+
     # Waveshare "C" 1024x600 7" Rev2.1 IPS capacitive touch (http://www.waveshare.com/7inch-HDMI-LCD-C.htm)
     if [ "${WAVESHARE_1024X600_C_2_1}" = "1" ]; then
         echo "# Waveshare \"C\" 1024x600 7\" Rev2.1 IPS capacitive touch screen" >> $CONFIG

```

agherzan, 2021-05-13:
Looks good to me. Could you create an MR for it?

sss22213, 2021-05-13:
hi, I create new PR for Raspberry Pi HD quality camera.

https://github.com/agherzan/meta-raspberrypi/pull/857

closed 2021-05-20 by sss22213