agherzan/meta-raspberrypi #864: Boot from USB on Pi 4 ?

om26er, 2021-05-26:
I have a Raspberry Pi on which Ubuntu and Raspbian OS boots fine from USB drive. I need to do a similar thing with my yocto based image. Has anyone made any progress on doing this ?

We want to flash our device's sdcard by first booting from USB and running a command to actually flash the sdcard. This will allow us to bootstrap our devices by booting a usb.

om26er, 2021-05-28:
I still haven't been able to boot from USB. However one thing is clear that the version of u-boot in dunfell branch won't really work. So I have currently ported latest u-boot recipe from poky master. Now u-boot at least detects the USB device.

om26er, 2021-05-28:
Here is part of the u-boot boot log
```
U-Boot 2021.04 (May 28 2021 - 01:03:52 +0000)

DRAM:  3.9 GiB
RPI 4 Model B (0xc03111)
MMC:   mmcnr@7e300000: 1, emmc2@7e340000: 0
Loading Environment from FAT... Card did not respond to voltage select! : -110
In:    serial
Out:   serial
Err:   serial
Net:   eth0: ethernet@7d580000
PCIe BRCM: link up, 5.0 Gbps x1 (SSC)
starting USB...
Bus xhci_pci: Register 5000420 NbrPorts 5
Starting the controller
USB XHCI 1.00
scanning bus xhci_pci for devices... 3 USB Device(s) found
       scanning usb for storage devices... 1 Storage Device(s) found
Hit any key to stop autoboot:  0
U-Boot> boot
Card did not respond to voltage select! : -110
Card did not respond to voltage select! : -110

Device 0: Vendor: Kingston Rev:  Prod: DataTraveler 3.0
            Type: Removable Hard Disk
            Capacity: 14755.2 MB = 14.4 GB (30218842 x 512)
... is now current device
Scanning usb 0:1...
Found U-Boot script /boot.scr
262 bytes read in 1 ms (255.9 KiB/s)
## Executing script at 02400000
Card did not respond to voltage select! : -110
Can't set block device
Card did not respond to voltage select! : -110
Saving Environment to FAT... Card did not respond to voltage select! : -110
Failed (1)
Bad Linux ARM64 Image magic!
SCRIPT FAILED: continuing...
libfdt fdt_check_header(): FDT_ERR_BADMAGIC
Card did not respond to voltage select! : -110
Scanning disk mmcnr@7e300000.blk...
Disk mmcnr@7e300000.blk not ready
Card did not respond to voltage select! : -110
Scanning disk emmc2@7e340000.blk...
Disk emmc2@7e340000.blk not ready
Scanning disk usb_mass_storage.lun0...
Found 3 disks
No EFI system partition
BootOrder not defined
EFI boot manager: Cannot load any image
```

agherzan, 2021-06-08:
The current setup of this BSP layer assumes SD boot. Check the `boot.scr` script. You would need to tweak it for USB boot: https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-bsp/rpi-u-boot-scr/files/boot.cmd.in

om26er, 2021-06-08:
@agherzan Thanks. I had figured it myself and now have USB boot working.

closed 2021-06-08 by om26er

rcla, 2021-06-15:
> I had figured it myself and now have USB boot working.

@om26er How did you do it?

I'm sure there is more than one person grateful to have Boot from USB on RPi4.

MartinLoeper, 2021-07-18:
@om26er I did not even manage to port u-boot from poky master to dunfell because we use meta-mender which adds complexity. This is just a nightmare. Hopefully, support for this feature is added in future yocto releases. We have the same usecase as the thread opener.

RaymiiOrg, 2022-03-04:
I also would like to know how you got usb boot working.

harunobukurokawa, 2023-05-05:
I succcess to boot yocto image from USB memory.

1. change  root=/dev/mmcblk0p2 to root=/dev/sda2 in cmdline.txt which is located in BOOT partition.
2.  modify boot.cmd.in file "mmc" to "usb" like as followings
<pre>
fatload usb 0:1 ${kernel_addr_r} Image
if test ! -e usb 0:1 uboot.env; then saveenv; fi;
booti ${kernel_addr_r} - ${fdt_addr}
</pre>
3.  execute the command
`mkimage -A arm64 -T script -C none -n "Boot script" -d "boot.cmd.in" boot.scr`
4. copy boot.scr file into USB memory (BOOT partition)