agherzan/meta-raspberrypi #868: RPI4 does not boot on dunfell branch  when uart_enable=0

hommeabeil, 2021-06-01:
**Description**

On the Dunfell branch the RPI4 does not boot if `uart_enable=0` is set. This was working properly when on the Zeus branch.

**Steps to reproduce the issue:**
1. Build the Dunfell branch.
2. Set `uart_enable=0` in your `config.txt`
3. Nothing will append, no kernel is booting.

**Describe the results you received:**

The RPI4 is not booting anymore. The kernel never get to log in tty1.

**Describe the results you expected:**

The board should boot as is would always do.

**Additional information you deem important (e.g. issue happens only occasionally):**

This was already tracked in the #733 pull request (which does not propose a proper fix). This looks like a U-Boot regression issue. More over, re-using the Zeus firmware  allows us to boot a kernel however, the mmc0 is never found so we can't go much further in the boot sequence.

hommeabeil, 2021-07-06:
More, on this, I had time to play a little more. It turns out that using the kernel 41.9 provided in this layer work perfectly well, no need to use the Zeus firmware or touch any other component, maybe it is related to the dtb ?

alex88, 2021-07-07:
Same happening here, did you try with 5.10 too? I see some files related to that as well

hommeabeil, 2021-07-09:
I start to look at the `.dts` in the kernel tree and I noticed this. In the kernel 4.19, the `/chosen` node does not contains a `stdout-path`, but in the kernel 5.4 the following node is present

```
chosen {
		/* 8250 auxiliary UART instead of pl011 */
		stdout-path = "serial1:115200n8";
	};
```

I am not an expert with those configuration, but this look promising. Does any one know what are the impact of `enable_uart=1` with the overlay ? Are they adding a value in the `chosen` node ? If they do this could explain it.

om26er, 2021-07-14:
@hommeabeil Can you share what error do you see in u-boot, can you clear your uboot.env and see if that helps ?

From what I can see the fdt_addr may change if you update the firmware, however since u-boot saves that in the env, the boot may break.

hommeabeil, 2021-07-15:
I can't see any error since uart1 (serial1) is disable. I do not think this come from the u-boot env since I can reproduce it from a fresh sd card.

@om26er  Do you have any clue on how to debug U-Boot without a serial port ?

alex88, 2021-07-15:
@om26er I usually build from scratch every time and I have the same issue, without a serial console I only see a black screen when powered

alex88, 2021-07-19:
There's no workaround that can be easily implemented at the moment right?
I really need to be able to use the main serial pins since the rpi we use have a RS485 connection on it and before going with another bootloader I want to make sure there is no other way to use uboot

alex88, 2021-07-19:
I was able to successfully boot by setting
```
# Enable UART
enable_uart=0
# Enable VC4 Graphics
# dtoverlay=vc4-fkms-v3d
dtoverlay=disable-bt
```

it seems there is no way to set a custom dtoverlay so I'll try to use a bbappend for that

agherzan, 2021-09-03:
The key information in the issue is that you're using uboot and it does look like this is a duplicate of https://github.com/agherzan/meta-raspberrypi/issues/732 so I'll close it and let the initial issue open.

closed 2021-09-03 by agherzan