agherzan/meta-raspberrypi #869: RPI4 CM4 cannot stream audio to hdmi0 audio output with gstreamer (audiotestsrc input)

nicolasb565, 2021-06-02:
**Description**
I cannot get gstreamer to stream to the hdmi0 audio output on RPI4 CM4 (with IO board) no matter what I tries.
I tried with hardknott and gatesgarth with latest code.
I am capable of reading EDID  using get-edid -b 11 
I tried using Inogeni 4kx, 4k2 (devices mades for capturing hdmi audio + video), also tried plugging on a nvidia rtx 2060.

The main problem I see in error logs from gstreamer is that it cannot query the formats for the alsa sink.
0:00:00.042798999  1186   0x55b152eaa0 ERROR                   alsa gstalsa.c:553:gst_alsa_probe_supported_formats:<alsasink0> failed to query formats

**Steps to reproduce the issue:**
Some settings in local.conf:
MACHINE = "raspberrypi4-64"
ENABLE_I2C = "1"
HDMI_FORCE_HOTPLUG = "1"
HDMI_MODE = "16"
HDMI_DRIVE = "2"
DISABLE_OVERSCAN = "1"
RPI_EXTRA_CONFIG  = "\ndtparam=audio=on\n"
KERNEL_MODULE_AUTOLOAD_rpi += "i2c-dev i2c-bcm2708"
VC4DTBO = "vc4-kms-v3d"

IMAGE_FEATURES += " ssh-server-openssh "
EXTRA_IMAGE_FEATURES += " debug-tweaks "

LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-omx commercial_gstreamer1.0-plugins-ugly commercial_ffmpeg commercial_mpeg2dec commercial_x264 commercial_gstreamer1.0-libav "
PACKAGECONFIG_append_pn-gstreamer1.0-plugins-bad = "kms"

DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"

IMAGE_INSTALL_append +=     " usbutils \
            v4l-utils \
            pciutils \
            ethtool \
            gstreamer1.0 \
            gstreamer1.0-meta-base \
            gstreamer1.0-plugins-bad \
            gstreamer1.0-plugins-base \
            gstreamer1.0-plugins-good \
            gstreamer1.0-plugins-ugly \
            gstreamer1.0-omx  \
            gstreamer1.0-vaapi \
            gstreamer1.0-libav \
            ffmpeg  \
            udev \
            libusb1 \
            kernel-modules \
            streamerapp \
            alsa-utils \
            read-edid \
            i2c-tools \
            swupdate \
            swupdate-www \
            "

1. Plug a device on the hdmi0 output that can capture audio from hdmi (tried many)
2. aplay -l to find the card number of vc4-hdmi-0
3. GST_DEBUG=2 gst-launch-1.0 audiotestsrc ! audioresample ! alsasink device=hw:cardNum,0
4. Also tried GST_DEBUG=2 gst-launch-1.0 audiotestsrc ! alsasink device=hw:cardNum,0

**Describe the results you received:**
root@raspberrypi4-64:~# GST_DEBUG=2 gst-launch-1.0 audiotestsrc ! audioresample ! alsasink device=hw:1,0
Setting pipeline to PAUSED ...
Pipeline is PREROLLING ...
0:00:00.042798999  1186   0x55b152eaa0 ERROR                   alsa gstalsa.c:553:gst_alsa_probe_supported_formats:<alsasink0> failed to query formats

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.687: gst_caps_intersect_full: assertion 'GST_IS_CAPS (caps2)' failed

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.687: gst_mini_object_unref: assertion 'mini_object != NULL' failed
0:00:00.043430658  1186   0x55b152eaa0 ERROR                   alsa gstalsa.c:553:gst_alsa_probe_supported_formats:<alsasink0> failed to query formats

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.687: gst_caps_intersect_full: assertion 'GST_IS_CAPS (caps2)' failed

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.687: gst_mini_object_unref: assertion 'mini_object != NULL' failed
0:00:00.043787062  1186   0x55b152eaa0 ERROR                   alsa gstalsa.c:553:gst_alsa_probe_supported_formats:<alsasink0> failed to query formats

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.688: gst_caps_intersect_full: assertion 'GST_IS_CAPS (caps2)' failed

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.688: gst_mini_object_unref: assertion 'mini_object != NULL' failed
0:00:00.045351711  1186   0x55b152eaa0 WARN         audio-resampler audio-resampler.c:274:convert_taps_gint16_c: can't find exact taps
0:00:00.045637911  1186   0x55b152eaa0 ERROR                   alsa gstalsa.c:553:gst_alsa_probe_supported_formats:<alsasink0> failed to query formats

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.690: gst_caps_intersect_full: assertion 'GST_IS_CAPS (caps2)' failed

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.690: gst_mini_object_unref: assertion 'mini_object != NULL' failed
0:00:00.045902890  1186   0x55b152eaa0 WARN                    alsa gstalsasink.c:467:set_hwparams:<alsasink0> Error -22 (Invalid argument) calling snd_pcm_hw_params_set_format (alsa->handle, params, alsa->format)
0:00:00.045935112  1186   0x55b152eaa0 WARN                    alsa gstalsasink.c:581:set_hwparams:<alsasink0> error: Sample format not available for playback: Invalid argument
ERROR: from element /GstPipeline:pipeline0/GstAlsaSink:alsasink0: Could not get/set settings from/on resource.
Additional debug info:
../gst-plugins-base-1.18.4/ext/alsa/gstalsasink.c(581): set_hwparams (): /GstPipeline:pipeline0/GstAlsaSink:alsasink0:
Sample format not available for playback: Invalid argument
ERROR: pipeline doesn't want to preroll.
0:00:00.046155424  1186   0x55b152eaa0 WARN                    alsa gstalsasink.c:902:gst_alsasink_prepare:<alsasink0> Error -22 (Invalid argument) calling set_hwparams (alsa)
0:00:00.046184683  1186   0x55b152eaa0 WARN                    alsa gstalsasink.c:947:gst_alsasink_prepare:<alsasink0> error: Setting of hwparams failed: Invalid argument
ERROR: from element /GstPipeline:pipeline0/GstAlsaSink:alsasink0: Could not get/set settings from/on resource.
Additional debug info:
../gst-plugins-base-1.18.4/ext/alsa/gstalsasink.c(947): gst_alsasink_prepare (): /GstPipeline:pipeline0/GstAlsaSink:alsasink0:
Setting of hwparams failed: Invalid argument
ERROR: pipeline doesn't want to preroll.
0:00:00.046442717  1186   0x55b152eaa0 ERROR                   alsa gstalsa.c:553:gst_alsa_probe_supported_formats:<alsasink0> failed to query formats

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.690: gst_caps_intersect_full: assertion 'GST_IS_CAPS (caps2)' failed

(gst-launch-1.0:1186): GStreamer-CRITICAL **: 20:50:08.690: gst_mini_object_unref: assertion 'mini_object != NULL' failed
Setting pipeline to NULL ...
0:00:00.046834953  1186   0x55b152eaa0 WARN           audiobasesink gstaudiobasesink.c:1191:gst_audio_base_sink_preroll:<alsasink0> error: sink not negotiated.
ERROR: from element /GstPipeline:pipeline0/GstAlsaSink:alsasink0: The stream is in the wrong format.
Additional debug info:
../gst-plugins-base-1.18.4/gst-libs/gst/audio/gstaudiobasesink.c(1191): gst_audio_base_sink_preroll (): /GstPipeline:pipeline0/GstAlsaSink:alsasink0:
sink not negotiated.
ERROR: pipeline doesn't want to preroll.
0:00:00.047208042  1186   0x55b152eaa0 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<audiotestsrc0> error: Internal data stream error.
0:00:00.047235375  1186   0x55b152eaa0 WARN                 basesrc gstbasesrc.c:3127:gst_base_src_loop:<audiotestsrc0> error: streaming stopped, reason not-negotiated (-4)
ERROR: from element /GstPipeline:pipeline0/GstAudioTestSrc:audiotestsrc0: Internal data stream error.
Additional debug info:
../gstreamer-1.18.4/libs/gst/base/gstbasesrc.c(3127): gst_base_src_loop (): /GstPipeline:pipeline0/GstAudioTestSrc:audiotestsrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Freeing pipeline ...

**Describe the results you expected:**
I expect it to be able to stream the audiotestsrc to hdmi0.

agherzan, 2021-11-14:
Hi @nicolasb565 . Thanks for your report. Sadly the logs above are all over the place. Could you please clean them up and provide the minimum set of configurations that reproduce the issue? Also, details on the revisions of the layers used are needed. Have you also tried to reproduce this issue on Raspbian?

nicolasb565, 2021-11-14:
We moved on from using vc4-kms-v3d because we need audio on the hdmi, and we also want tvservice and edid probing to works.

I would just test it on latest version of yocto for raspberry pi 4 or raspbian, making sure to use vc4-kms-v3d instead of vc4-fkms-v3d. Try to see if the audio works on the hdmi output when using it in a gstreamer pipeline.

The following issues are for vc4-fkms-v3d with latest version of yocto on raspberry pi 4:

1- Hdmi hotplug detection not working properly if no hdmi device is connected at boot. Basically tvservice does not see the hotplug in that case. This is a more important issue for us, and I read on raspberry pi forum that it date way back to older devices. We would like to be able to detect hotplug, and then set prefered hdmi timing with tvservice. Even setting prefered timing does not work because tvservice just can't figure out what it is. Edid probing always return that of the first device connected on hdmi when nothing was plugged at boot on hdmi, even if you plug other monitors.

2- Tvservice can change hdmi timing, but it does not change the framebuffer, and fbset is equally useless at changing the framebuffer. Need to use tricks to get around having framebuffer of incorrect size so that we are able to change resolution without a reboot.