agherzan/meta-raspberrypi #871: WPE DRM not displaying anything on RPI4 while works on RPI3

nandra, 2021-06-08:
I'm trying to build image with wpewebkit by using meta-webkit. I'm using hardknott release and image build smoothly. 
I'm trying to use opensource VC4 + DRM backend to display web content following this instructions: https://github.com/Igalia/meta-webkit/wiki/RPi#option-2-building-an-image-with-mesa-vc4-driver-weston-and-wpebackend-fdo

While I build it for rpi3 I can display webpages etc. but image build for rpi4 just don't display anything at all. Just want to ask if anyone have experience with that. 

I was also looking at https://github.com/WebPlatformForEmbedded/meta-wpe layer but it seems it's use weston + wayland. I want to use only DRM and not window manager. Thanks for any info.

agherzan, 2021-06-08:
I also only tested this back in the days with RPi3. I haven't tried it on RPi4 at all. We'd need to look into it a bit.

nandra, 2021-06-08:
OK thanks if I can help somehow with testing or so pls let me know.

kraj, 2021-06-12:
are you able to run graphics at all ? e.g. does weston render ok on rpi4 ? and did you try 32bit rpi4 or 64bit ? I would suggest to use 64bit.

nandra, 2021-06-12:
Yes. With weston it works, but I would like to use drm backend cos I don't need window manager at all. Using 32bit only.Thanks.

kraj, 2021-06-12:
can you try 64bit?

nandra, 2021-06-14:
Tried and there is build error for cairo: 
```
/home/builder/build/tmp/work/cortexa72-poky-linux/cairo/1.16.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/10.2.0/ld: cannot find -lbrcmEGL
| /home/builder/build/tmp/work/cortexa72-poky-linux/cairo/1.16.0-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/../../libexec/aarch64-poky-linux/gcc/aarch64-poky-linux/10.2.0/ld: cannot find -lbrcmGLESv2

```

kraj, 2021-06-14:
you seem to be using userland graphics. which wont work in 64bit

nandra, 2021-06-16:
Tried wpewebkit from meta-webkit for 64 bit and same issue. cog seems to start but it reports:
```
cog --platform drm https://google.sk
Cog-Message: 06:59:02.651: <https://google.sk/> Load started.
Cog-Message: 06:59:03.114: <https://www.google.sk/> Redirected.
Cog-Message: 06:59:03.488: <https://www.google.sk/> Loading...

** (cog:274): WARNING **: 06:59:05.180: failed to set mode: Invalid argument
Cog-Message: 06:59:05.393: <https://www.google.sk/> Loaded successfully.
```

but display is empty.

nandra, 2021-06-16:
Tried 64bit wpe-westeros-image raspberrypi4-64 but similar issue. On control webpage I was not able to start webkit:

[image]

and wpeframework.service displays only:
```
Jun 16 11:42:46 raspberrypi4-64 WPEFramework[352]: 1623843766280: INFO: ../git/drm/westeros-gl/westeros-gl.c:2870 opened /dev/d
ri/card0: master 1
Jun 16 11:42:46 raspberrypi4-64 WPEFramework[352]: 1623843766280: INFO: ../git/drm/westeros-gl/westeros-gl.c:2917 westeros-gl: 
have drm atomic mode setting
Jun 16 11:42:46 raspberrypi4-64 WPEFramework[352]: 1623843766312: INFO: ../git/drm/westeros-gl/westeros-gl.c:3008 wstInitCtx: c
urrent mode 640x480@60
Jun 16 11:42:46 raspberrypi4-64 WPEFramework[352]: 1623843766317: INFO: ../git/drm/westeros-gl/westeros-gl.c:3204 wstInitCtx; f
ound 7 overlay planes
```

So it seems it is completely broken ;).

agherzan, 2021-10-30:
@nandra did you manage to figure this one out? Maybe the wpe guys would know more?

closed 2021-11-14 by agherzan