agherzan/meta-raspberrypi #875: CM4 change console port in u-boot

nandra, 2021-06-22:
Hi,

I'm trying to change default serial console port on cm4 from ttyS0 to ttyAMA1 (UART2). I've configured:
```
RPI_KERNEL_DEVICETREE_OVERLAYS_append = " overlays/uart2.dtbo"
SERIAL_CONSOLES = "115200;ttyAMA1"
```
+ added entry to config.txt : 
```
dtoverlay=uart2
``` and linux console is redirected properly to that UART. Only issue I have is u-boot. I've tried to update kernel dts :
```
diff --git a/arch/arm/boot/dts/bcm2711-rpi-4-b.dts b/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
index 21b20e3..e955044 100644
--- a/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
+++ b/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
@@ -211,6 +211,12 @@
        status = "okay";
 };
 
+&uart2 {
+       pinctrl-names = "default";
+       pinctrl-0 = <&uart2_pins>;
+       status = "okay";
+};
+
 &vchiq {
        interrupts = <GIC_SPI 34 IRQ_TYPE_LEVEL_HIGH>;
 };
@@ -239,7 +245,7 @@
 
        aliases {
                serial0 = &uart1;
-               serial1 = &uart0;
+               serial1 = &uart2;
                mmc0 = &emmc2;
                mmc1 = &mmcnr;
                mmc2 = &sdhost;

```

but I'm not sure because u-boot still print to ttyS0. Does anybody have some experience with that? Looked at u-boot source (using dunfell 2020.01) looks like model is not properly detected due missing entry so I just overtake it from later u-boot release and board is now properly identified with:
```
+       [0x14] = {
+               "Compute Module 4",
+               DTB_DIR "bcm2711-rpi-cm4.dtb",
+               true,
+       },
```

but file bcm2711-rpi-cm4.dtb is mising even in mainline u-boot. Does u-boot use kernel devicetree or I'm completely on wrong track? Thanks.

nandra, 2021-07-14:
Any ideas? Thanks.

agherzan, 2021-09-03:
Hi, @nandra did you manage to figure this one out. I remember seeing some hardcoded assumptions on the console in the uboot code when working for the RPI4 port.

closed 2021-11-14 by agherzan