agherzan/meta-raspberrypi #876: CM3 U-Boot patch does not apply on Hardknott branch

Metatron2k2, 2021-07-06:
**Description**

U-Boot patch for CM3 does not apply on hardknott.

**Steps to reproduce the issue:**
1. Set Machine to CM3
2. Build image depending on uboot

**Describe the results you received:**
Error that hunk 2 patch does not apply. 
ERROR: u-boot-1_2021.01-r0 do_patch: Command Error: 'quilt --quiltrc /bsp/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/u-boot/1_2021.01-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
stdout: Applying patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch
patching file drivers/core/device.c
Hunk #1 succeeded at 456 (offset 81 lines).
Hunk #2 FAILED at 418.
1 out of 2 hunks FAILED -- rejects in file drivers/core/device.c
Patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch does not apply (enforce with -f)

**Describe the results you expected:**
Patch should apply and no error should be throwb.

**Additional information you deem important (e.g. issue happens only occasionally):**
The error occurs for me trying to build an swupdate-image (meta-swupdate)

**Additional details (revisions used, host distro, etc.):**
Host is a nspawn container based on Debian 10 (stable)

kraj, 2021-07-06:
@Metatron2k2 I hope you are on same branch ( hardknott ) of other layers too. Secondly, I would suggest to disable all layers except core layer and meta-rpi and run `bitbake -cpatch u-boot` if it still fails then we can perhaps update/refresh the patch and send a PR here, but if its conflicting with another patch from different layer then perhaps we have to find an amicable way to solve this involving contending layers. Perhaps move the patch to oe-core or something so the apply order is always same.

Metatron2k2, 2021-07-07:
@kraj Yes everything was in in hardknott branch. I removed all non-openembedded and non-poky layers. Result is still the same
```
bitbake -cpatch u-boot
Loading cache: 100% |                                                                                                                                                                            | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |###########################################################################################################################################################################| Time: 0:00:50
Parsing of 2494 .bb files complete (0 cached, 2494 parsed). 3836 targets, 389 skipped, 0 masked, 0 errors.
Removing 1 recipes from the allarch sysroot: 100% |###############################################################################################################################################| Time: 0:00:00
Removing 23 recipes from the cortexa7t2hf-neon-vfpv4 sysroot: 100% |##############################################################################################################################| Time: 0:00:08
Removing 24 recipes from the raspberrypi_cm3 sysroot: 100% |######################################################################################################################################| Time: 0:00:01
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi-cm3"
DISTRO               = "poky"
DISTRO_VERSION       = "3.3.1"
TUNE_FEATURES        = "arm vfp cortexa7 neon vfpv4 thumb callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "hardknott:60d51ca484e43f4ef38f313956e510c25dcd7dac"
meta-oe
meta-multimedia
meta-networking
meta-python
meta-filesystems
meta-perl            = "hardknott:c51e79dd854460c6f6949a187970d05362152e84"
meta-raspberrypi     = "hardknott:064f5404ea90f02bd15088de6317692098d9f770"

Initialising tasks: 100% |########################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Local 0 Network 0 Missed 0 Current 20 (0% match, 100% complete)
NOTE: Executing Tasks
ERROR: u-boot-1_2021.01-r0 do_patch: Command Error: 'quilt --quiltrc /bsp/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/u-boot/1_2021.01-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
stdout: Applying patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch
patching file drivers/core/device.c
Hunk #1 succeeded at 456 (offset 81 lines).
Hunk #2 FAILED at 418.
1 out of 2 hunks FAILED -- rejects in file drivers/core/device.c
Patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch does not apply (enforce with -f)

stderr:
ERROR: Logfile of failure stored in: /bsp/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/u-boot/1_2021.01-r0/temp/log.do_patch.3948502
ERROR: Task (/home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/u-boot_2021.01.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 93 tasks of which 92 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/u-boot_2021.01.bb:do_patch
Summary: There was 1 ERROR message shown, returning a non-zero exit code.
```
Bitbake log:
```
cat /bsp/tmp/work/raspberrypi_cm3-poky-linux-gnueabi/u-boot/1_2021.01-r0/temp/log.do_patch.3948502
DEBUG: Executing python function extend_recipe_sysroot
NOTE: Direct dependencies are ['/home/muellerm/yocto-build-scripts/poky/meta/recipes-devtools/quilt/quilt-native_0.66.bb:do_populate_sysroot', 'virtual:native:/home/muellerm/yocto-build-scripts/poky/meta/recipes-devtools/patch/patch_2.7.6.bb:do_populate_sysroot']
NOTE: Installed into sysroot: []
NOTE: Skipping as already exists in sysroot: ['quilt-native', 'patch-native', 'autoconf-native', 'attr-native', 'libtool-native', 'automake-native', 'm4-native', 'texinfo-dummy-native', 'gnu-config-native', 'gettext-minimal-native']
DEBUG: Python function extend_recipe_sysroot finished
DEBUG: Executing python function do_patch
DEBUG: Executing python function patch_do_patch
DEBUG: Searching for 0001-add-valid-fdt-check.patch in paths:
    /home/muellerm/yocto-build-scripts/poky/meta-raspberrypi/recipes-bsp/u-boot/files/poky
    /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/u-boot-2021.01/poky
    /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/u-boot/poky
    /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/files/poky
    /home/muellerm/yocto-build-scripts/poky/meta-raspberrypi/recipes-bsp/u-boot/files/raspberrypi-cm3
    /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/u-boot-2021.01/raspberrypi-cm3
[... 240 lines cut ...]
    /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/u-boot-2021.01/
    /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/u-boot/
    /home/muellerm/yocto-build-scripts/poky/meta/recipes-bsp/u-boot/files/
NOTE: Applying patch '0001-add-valid-fdt-check.patch' (../poky/meta/recipes-bsp/u-boot/files/0001-add-valid-fdt-check.patch)
NOTE: Applying patch 'CVE-2021-27097-1.patch' (../poky/meta/recipes-bsp/u-boot/files/CVE-2021-27097-1.patch)
[... cut ...]
stdout: Applying patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch
patching file drivers/core/device.c
Hunk #1 succeeded at 456 (offset 81 lines).
Hunk #2 FAILED at 418.
1 out of 2 hunks FAILED -- rejects in file drivers/core/device.c
Patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch does not apply (enforce with -f)

stderr:
DEBUG: Python function patch_do_patch finished
DEBUG: Python function do_patch finished
```
For me it does not seem like any patch not part of open-embedded or poky being applied. Any Idea why this happens?

kraj, 2021-07-07:
thanks, this means this patch needs to be forward ported as it seems it was missed when oe-core u-boot got updated.

Metatron2k2, 2021-07-13:
HI,
qho would be the right person to ask about this. I don't know the insights of uboot and I don't really understand what the patch is doing.
Best

emilm, 2021-07-28:
> HI,
> qho would be the right person to ask about this. I don't know the insights of uboot and I don't really understand what the patch is doing.
> Best

It moves the platform data translation more early in the file so it executes in an order that makes cm3 boot
https://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/tree/recipes-bsp/u-boot/files/0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch

Metatron2k2, 2021-08-03:
Hi  @emilm,

I‘ve seen you already created a patch on the dunfell branch. Would you be so kind to share it with me?

Best,
Stefan

emilm, 2021-08-03:
Hello @Metatron2k2  ! Does the patch break for you the same way?

Metatron2k2, 2021-08-03:
Hey, I just double checked. In fact it breaks in a different way. I‘m also not sure if it is really necessary here though….
I think I‘ll just remove it and see if the CM3 successfully  boots
.

Metatron2k2, 2021-08-05:
So it turns out that it does not boot. It seems like the file changed quite a bit and I'm not sure where to start. Some help would be appreciated.

emilm, 2021-08-05:
If you are on Hardknott, my patch for Dunfell would probably not work.
I am not on Hardknott, so you have to tell me which git revision the u-boot is on, so I can view the offending file.

Metatron2k2, 2021-08-05:
Hey @emilm,
the git rev is c4fddedc48f336eabc4ce3f74940e6aa372de18c .

Looking at the files it seems like the order is correct (but I'm not 100% sure). On the CMIO board i get a few blinks on the activity LED and then nothing. Debug UART (I assume GPIO 14 and 15? ) is also not doing anythin.

Metatron2k2, 2021-08-05:
Looking at the git history upstream it seems like the patch should already be applied. But i still don't get any debut output or anything useful from uboot. Any idea?

emilm, 2021-08-05:
Seems like the right order. However u-boot has been refactored several times.
https://source.denx.de/u-boot/u-boot/-/blob/c4fddedc48f336eabc4ce3f74940e6aa372de18c/drivers/core/device.c#L439

[image]

The patch:
https://github.com/agherzan/meta-raspberrypi/blob/hardknott/recipes-bsp/u-boot/files/0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch

It seems outdated. 

No other ideas than comparing a patched u-boot from Dunfell with this one and see how the code ultimately ends up working in sequence :/

Metatron2k2, 2021-08-05:
We're only using ENABLE_UART=1 in the config no other overlays. Strangly enough i do not get anything on the UART. The UART adapter itself seems fine (at least it is working on my PI4). Any Idea why I do not get anything from u-boot on uart at all?

emilm, 2021-08-05:
What pins is it wired up to, and what is your cmdline.txt?

Metatron2k2, 2021-08-05:
cmdline.txt:
`dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait logo.nologo`
GPIO14(TX) is connected to RX of my USB UART and GPIO15(RX) to TX. GND to GND

emilm, 2021-08-05:
Did you try pins 32 and 33, the UARTs are flipped in CM3 also afaik.

Metatron2k2, 2021-08-05:
Yeah tried both. Also tried a pi zero W as UART adapter. I was able to get output on our "old" production image.

I'd like to run a build against dunfell to see if it is working there. Could you supply me with your patch so I can get U-Boot up and running there?

Once i know the old U-Boot is working well, I would try to get some more support from the U-Boot mailing list.

emilm, 2021-08-05:
OK, you should try without the patch first.
 Without overlays it should actually boot without patch. Did for me.

[0001-Move-ofdata_to_platdata-earlier-CM3-fix.patch.txt](https://github.com/agherzan/meta-raspberrypi/files/6942001/0001-Move-ofdata_to_platdata-earlier-CM3-fix.patch.txt)

Metatron2k2, 2021-08-06:
Hey @emilm Thanks for the patch. I can now confirm that i get Debug output from U-Boot 2020.1 without the patch. With your patch it boots just fine.

I assume that U-Boot in Hardknott is broken for CM3 completely.

Metatron2k2, 2021-08-06:
Ok, as it is not an issue with the patch I've opened a new issue #897

agherzan, 2021-11-14:
Closing this issue in favour of https://github.com/agherzan/meta-raspberrypi/issues/897

closed 2021-11-14 by agherzan