agherzan/meta-raspberrypi #878: Dunfell is using old firmware

om26er, 2021-07-08:
It seems the dunfell branch is using a very old firmware version. That firmware does not support the new BCM-USB-MSD boot that came with Raspberry Pi4 (CM4/Pi400).

Fortuntely, it seems that updating the firmware is a very simple task and below patch worked well for me and solved the issue I was facing.
```
diff --git a/recipes-bsp/common/raspberrypi-firmware.inc b/recipes-bsp/common/raspberrypi-firmware.inc
index 32cc480..41a9fd8 100644
--- a/recipes-bsp/common/raspberrypi-firmware.inc
+++ b/recipes-bsp/common/raspberrypi-firmware.inc
@@ -1,9 +1,10 @@
-RPIFW_DATE ?= "20201002"
-SRCREV ?= "11e3c314bc2b64f7d862bac00ff3d9f42f3c5a50"
+RPIFW_DATE ?= "20210527"
+SRCREV ?= "1eb8921ae7158249894f89e8bb6c4bfb04ec7377"
 RPIFW_SRC_URI ?= "https://github.com/raspberrypi/firmware/archive/${SRCREV}.tar.gz;downloadfilename=raspberrypi-firmware-${SRCREV}.tar.gz"
 RPIFW_S ?= "${WORKDIR}/firmware-${SRCREV}"
 
 SRC_URI = "${RPIFW_SRC_URI}"
-SRC_URI[sha256sum] = "b8c9f5a3e987f418f11526cbf59f6a6c8103e8f5e94bccffb9ef3fd7ae834f98"
+SRC_URI[sha256sum] = "32fb5cc72703247622839c3cb4df2f369f91d2185770df867ac9758283bf5380"
+
 
 PV = "${RPIFW_DATE}"
```

Is there a reason why an old firmware version is being used, if none, shall I propose a PR with the above patch ?

kraj, 2021-07-08:
thanks @om26er can you cook up a patch and post a pull request

om26er, 2021-07-08:
@kraj created a pull request, please check https://github.com/agherzan/meta-raspberrypi/pull/879

agherzan, 2021-07-08:
Let's first land an update for master and then cherry-pick it and PR it to dunfell.

PR #880 (2021-07-09): Update raspberrypi firmware to 20210527
Update raspberrypi firmware to 20210527

Fixes #878
closed 2021-07-09 by agherzan