agherzan/meta-raspberrypi #885: 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch does not apply correctly

emilm, 2021-07-28:
**Description**

yocto fails to apply patch

**Steps to reproduce the issue:**
1. use machine raspberrypi-cm3
2. bitbake -c configure u-boot 

**Describe the results you received:**

```
ERROR: u-boot-1_2020.01-r0 do_patch: Command Error: 'quilt --quiltrc [...]/u-boot/1_2020.01-r0/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch
can't find file to patch at input line 21
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From 336d86ebd146905cf4384912f4f27699b6e37c72 Mon Sep 17 00:00:00 2001
|From: Simon Glass <sjg@chromium.org>
|Date: Sun, 29 Dec 2019 21:19:17 -0700
|Subject: [PATCH] dm: core: Move ofdata_to_platdata() call earlier
|
|This method is supposed to extract platform data from the device tree. It
|should be done before the device itself is probed. Move it earlier in the
|device_probe() function.
|
|Upstream-Status: Backport
|
|Signed-off-by: Simon Glass <sjg@chromium.org>
|---
| drivers/core/device.c | 14 +++++++-------
| 1 file changed, 7 insertions(+), 7 deletions(-)
|
|diff --git a/drivers/core/device.c b/drivers/core/device.c
|index 4e03708..291ff4c 100644
|--- a/drivers/core/device.c
|+++ b/drivers/core/device.c
--------------------------
No file to patch.  Skipping patch.
2 out of 2 hunks ignored
Patch 0001-dm-core-Move-ofdata_to_platdata-call-earlier.patch does not apply (enforce with -f)
````

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

u-boot commit c66dbbe07e5da75c4704aa9b28036e4752814c22
meta-raspberry commit 59c2d6f7a8b1239bd7b587b9180c2a55f9c695a2
using Dunfell

I could propose a PR for working patch since i had to do this myself with devtool modify.

agherzan, 2021-09-03:
Sounds good. Feel free to push a PR.

Mastiggia, 2022-01-05:
Hello,

I can't reproduce this issue with:
- poky: dunfell
- meta-openembedded: dunfell
- meta-raspberrypi: dunfell
- MACHINE = "raspberrypi-cm3"

Can we maybe close it?

agherzan, 2022-01-10:
Thanks for confirming @Mastiggia. Closed.

closed 2022-01-10 by agherzan