agherzan/meta-raspberrypi #897: [CM3+/Hardknott] U-Boot is defunct

Metatron2k2, 2021-08-06:
**Description**
U-Boot of Hardknott Branch does not boot on CM3+ at all. U-Boot 2020.1 from Dunfell works as expected (after updating with @emilm 's Patch.

Setting uart_2ndstage=1 in config.txt reveals that the system stops after loading U-Boot

**Steps to reproduce the issue:**
1. Build core-image-minimal (with U-Boot) on hardknott
2. Write Image to CM3+
3. Try to boot.

**Describe the results you received:**
Stuck on boot.

**Describe the results you expected:**
Booting system.

**Additional information you deem important (e.g. issue happens only occasionally):**
2nd stage logs:

```
Raspberry Pi Bootcode
Read File: config.txt, 36315
Read File: start_x.elf, 3707432 (bytes)
Read File: fixup_x.dat, 10297 (bytes)
MESS:00:00:03.831439:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.853904:0: brfs: File read: 36315 bytes
MESS:00:00:03.862208:0: brfs: File read: /mfs/sd/dt-blob.bin
MESS:00:00:03.888049:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:03.894214:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:03.900551:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:03.906888:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:03.913225:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:03.919562:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:03.925898:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:03.932235:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:03.938572:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:03.944909:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:03.951005:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:03.956206:0: brfs: File read: 53198 bytes
MESS:00:00:03.961998:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:03.984599:0: gpioman: gpioman_get_pin_num: pin DISPLAY_SDA not defined
MESS:00:00:04.005369:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:04.011427:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:04.019173:0: gpioman: gpioman_get_pin_num: pin FLASH_0_ENABLE not defined
MESS:00:00:04.026617:0: gpioman: gpioman_get_pin_num: pin FLASH_0_INDICATOR not defined
MESS:00:00:04.034385:0: gpioman: gpioman_get_pin_num: pin BT_ON not defined
MESS:00:00:04.041020:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:04.224410:0: *** Restart logging
MESS:00:00:04.226904:0: brfs: File read: 36315 bytes
MESS:00:00:04.234264:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.240947:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.247806:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.254663:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:04.261521:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:04.268379:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:04.275235:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:04.282094:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:04.288951:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:04.295809:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:04.302425:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:04.308326:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:04.316121:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:04.322978:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:04.329837:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:04.336693:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:04.343552:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:04.350410:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:04.357266:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:04.364125:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:04.370983:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:04.377598:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:04.383471:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:04.391958:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:04.397653:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
MESS:00:00:04.414661:0: gpioman: gpioman_get_pin_num: pin CAMERA_0_I2C_PORT not defined
MESS:00:00:04.421305:0: dtb_file 'bcm2710-rpi-cm3.dtb'
MESS:00:00:04.430166:0: brfs: File read: /mfs/sd/bcm2710-rpi-cm3.dtb
MESS:00:00:04.434827:0: Loading 'bcm2710-rpi-cm3.dtb' to 0x100 size 0x690a
MESS:00:00:04.454150:0: brfs: File read: 26890 bytes
MESS:00:00:04.459742:0: brfs: File read: /mfs/sd/overlays/overlay_map.dtb
MESS:00:00:04.527818:0: brfs: File read: 1559 bytes
MESS:00:00:04.532218:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:04.557640:0: dtparam: i2c1=on
MESS:00:00:04.567676:0: dtparam: i2c_arm=on
MESS:00:00:04.578515:0: brfs: File read: 36315 bytes
MESS:00:00:04.584389:0: brfs: File read: /mfs/sd/overlays/vc4-kms-v3d.dtbo
MESS:00:00:04.633411:0: Loaded overlay 'vc4-kms-v3d'
MESS:00:00:04.738389:0: brfs: File read: 2721 bytes
MESS:00:00:04.743858:0: brfs: File read: /mfs/sd/overlays/disable-bt.dtbo
MESS:00:00:04.757248:0: dterror: can't find symbol 'uart0_pins'
MESS:00:00:04.761468:0: Failed to resolve overlay 'disable-bt'
MESS:00:00:04.772582:0: brfs: File read: 1073 bytes
MESS:00:00:04.776240:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:04.781163:0: Read command line from file 'cmdline.txt':
MESS:00:00:04.787040:0: 'dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait logo.nologo'
MESS:00:00:06.569069:0: gpioman: gpioman_get_pin_num: pin WL_ON not defined
MESS:00:00:06.578513:0: brfs: File read: 101 bytes
MESS:00:00:06.628147:0: brfs: File read: /mfs/sd/kernel7.img
MESS:00:00:06.632108:0: Loading 'kernel7.img' to 0x8000 size 0x7ae04
MESS:00:00:06.638191:0: Device tree loaded to 0x2eff9100 (size 0x6eb5)
MESS:00:00:06.645854:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:06.652149:0: uart: Baud rate change done...
MESS:00:00:06.655577:0: uart: Baud rate change done...
MESS:00:00:06.662784:0: gpioman: gpioman_get_pin_num: pin SDCARD_CONTROL_POWER not defined
```
**Additional details (revisions used, host distro, etc.):**
Poky Hardknott with meta-raspberry

jsebrd, 2022-05-11:
This is also an issue in the `kirkstone` branch for `CM3` builds.  I can provide more details if desired.

jsebrd, 2022-05-12:
I can also confirm I cannot get u-boot to work with `CM3` or `CM3+` using current `kirkstone` branch commits or when trying latest `master` branch.  I can provide additional details if wanted.  I have also generated a patch against the commit to remove the u-boot patch file for `CM3` builds, but again no luck.

MartinLoeper, 2022-07-03:
Guys, I do not know if this is the same issue I was facing, but u-boot did not work for me on kirkstone and I did the following to solve it:

**Modify config.txt as follows:**

- remove all the unnecessary comments from config.txt (see explanation in https://github.com/raspberrypi/firmware/issues/1575#issuecomment-834790815) to strip it down to only a few lines of actual config
- set `dtoverlay=disable-bt` to be able to use the UART properly for anything following after the second stage bootloader finished

Z-Aurora99, 2025-09-08:
Hi,
I have the same issue when I build based on **scarthgap** with 32bit u-boot for CM3.
but it's ok based on **walnascar**.

the config.txt:

# Console UART
enable_uart=1
uart_2ndstage=1
dtoverlay=uart0
dtoverlay=disable-bt