agherzan/meta-raspberrypi #907: Unable to play QML video on raspberrypi cm3 on warrior branch

Shravansingh, 2021-08-24:
**Description**

I am trying to use qt multimedia to play an MP4 video on raspberrypi cm3 on warrior branch. I am able to play the said video using OMX player but not with Qt gstreamer.

**Describe the results you received:**
On the Qml side: I get the following error

```
qt.multimedia.video: source is QDeclarativeAudio(0x1bffe60)
qt.multimedia.video: media object is QMediaPlayer(0x1c8c530)
qt.multimedia.video: found videonode plugin "egl" 0x1efddc8
qt.multimedia.video: updatePaintNode: no frames yet
qt.multimedia.video: updatePaintNode: no frames yet
Warning: "Failed to connect: Connection refused"
qt.multimedia.video: Video surface format: QVideoSurfaceFormat(Format_YUV420P, QSize(640, 480), viewport=QRect(0,0 640x480), pixelAspectRatio=QSize(1, 1), handleType=NoHandle, yCbCrColorSpace=YCbCr_Undefined)
    handleType = QVariant(QAbstractVideoBuffer::HandleType, )
    pixelFormat = QVariant(QVideoFrame::PixelFormat, )
    frameSize = QVariant(QSize, QSize(640, 480))
    frameWidth = QVariant(int, 640)
    viewport = QVariant(QRect, QRect(0,0 640x480))
    scanLineDirection = QVariant(QVideoSurfaceFormat::Direction, )
    frameRate = QVariant(double, 25)
    pixelAspectRatio = QVariant(QSize, QSize(1, 1))
    sizeHint = QVariant(QSize, QSize(640, 480))
    yCbCrColorSpace = QVariant(QVideoSurfaceFormat::YCbCrColorSpace, )
    mirrored = QVariant(bool, false) all supported formats: (Format_YUV420P, Format_YV12, Format_NV12, Format_NV21, Format_UYVY, Format_YUYV, Format_RGB32, Format_ARGB32, Format_BGR32, Format_BGRA32, Format_RGB565)
```

When I run the dmesg I get the bellow error 
```
[21425.092624] cma: cma_alloc: alloc failed, req-size: 113 pages, ret: -12
[21425.092647] bcm2835-codec bcm2835-codec: dma_alloc_coherent of size 462848 failed
[21425.100286] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf cfd0b4e2
[21425.100413] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf a39e1a2a
[21425.100529] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 7639b224
[21425.100637] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 735ec6a6
[21425.100747] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 46ccd78a
[21425.100857] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 0e6f3a19
[21425.100993] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 0f342248
[21425.101114] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 26c74db6
[21425.101224] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf ed904bf4
[21425.101333] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 15f5ce9b
[21425.101442] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 26246ec7
[21425.101922] bcm2835_mmal_vchiq: buffer_from_host: msg_context not allocated, buf 6ab472b7
```

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**
Do I need to incorporate this bellow mentioned ticket to get rid of cma_alloc issue
https://github.com/raspberrypi/linux/issues/2680#issue-359609339

**Additional details (revisions used, host distro, etc.):**

My Local.conf 
DISTRO_FEATURES_append = " usbhost ppp systemd wifi pulseaudio alsa opengl"
DISTRO_FEATURES_remove = " wayland"
MACHINE_FEATURES_remove = " vc4graphics"
PACKAGECONFIG_append_pn-qtmultimedia = " gstreamer alsa"
IMAGE_INSTALL_append = " ffmpeg gstreamer1.0 gstreamer1.0-plugins-base gstreamer1.0-libav gstreamer1.0-rtsp-server gstreamer1.0-meta-base gstreamer1.0-plugins-ugly gstreamer1.0-plugins-good gst-examples pulseaudio alsa-utils alsa-plugins omxplayer"

agherzan, 2021-08-25:
Can you try it on master?

agherzan, 2021-09-03:
The reason is that `warrior` is not a branch that we maintain. So I think this needs a bit of investigation and the first test I'd do is to see if it is reproducible on master.

closed 2021-11-14 by agherzan