agherzan/meta-raspberrypi #908: Raspberry pi 4b (including rev 1.4) not booting on yocto dunfell

kmmenna, 2021-08-31:
**Description**

I'm trying to build a custom image with Yocto Dunfell to raspberry pi 4b(2GB), all build runs fine but when a burn the SD card, the rpi4 don't boot.

**Steps to reproduce the issue:**
1. checkout poky and all layers(below) in dunfell version
2. configure mender and build a custom image based on: core-image-base
3. burn SD card and try boot the rpi4

**Describe the results you received:**
No output image on HDMI, just turn on power led and 3 short blinks green led

**Describe the results you expected:**
Boot the system on rpi4

**Additional details (revisions used, host distro, etc.):**
Doesn't work on 2019 rpi4 model and rev 1.4 too.
The same image works fine in raspberry pi 3b.
poky: dunfell
meta-raspberrypi: dunfell
meta-mender: dunfell

**Layers in use**:
```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly

POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  ${TOPDIR}/../poky/meta \
  ${TOPDIR}/../poky/meta-poky \
  ${TOPDIR}/../poky/meta-yocto-bsp \
  ${TOPDIR}/../poky/meta-intel \
  ${TOPDIR}/../poky/openembedded-core/meta \
  ${TOPDIR}/../poky/meta-openembedded/meta-oe \
  ${TOPDIR}/../poky/meta-openembedded/meta-networking \
  ${TOPDIR}/../poky/meta-openembedded/meta-multimedia \
  ${TOPDIR}/../poky/meta-openembedded/meta-perl \
  ${TOPDIR}/../poky/meta-openembedded/meta-python \  
  ${TOPDIR}/../poky/meta-clang \
  ${TOPDIR}/../poky/meta-python2 \
  ${TOPDIR}/../poky/meta-security \
  ${TOPDIR}/../poky/meta-raspberrypi \
  ${TOPDIR}/../poky/meta-intel-realsense \
  ${TOPDIR}/../poky/meta-mender/meta-mender-core \
  ${TOPDIR}/../poky/meta-mender/meta-mender-raspberrypi \
"
```

kraj, 2021-09-01:
can you hook up serial console and see if you can find out something on it ?

agherzan, 2021-09-03:
Can you also share the content of your `cmdline.txt` file in the root of the boot partition?

kmmenna, 2021-09-04:
Hi, the cmdline.txt of my boot partition is:
```
dwc_otg.lpm_enable=0   rootfstype=ext4 rootwait   logo.nologo  root=${mender_kernel_root}
```
I will try to get console output but a I will need tools for that , so it may take a while.

Thanks for help!

agherzan, 2021-09-07:
Could you try to reproduce the issue with your mender integration? Only using oe-core and meta-raspberrypi.

kmmenna, 2021-09-09:
Sure, I will generate a entire new image with this configuration and return with the result.

kmmenna, 2021-10-20:
Hi I had tested using only the meta-raspberrypi layer and the issue didn't happen. Closing the issue.

closed 2021-10-20 by kmmenna