agherzan/meta-raspberrypi #909: Raspberrypi4 WiFi is not working using core-image-full-cmdline

gitgonewithwind, 2021-08-31:
sorry to reopen this issue:
https://github.com/agherzan/meta-raspberrypi/issues/841
 I have the some issue with core-image-full-cmdline ?
I do not know how to enable the wifi work.
before I do not use the pi 4 wifi function with core-image-full-cmdline, but one of our projects will use it
any help?
I have to switch to core-image-base?
what different between core-image-full-cmdline and core-image-base ?
any help?

agherzan, 2021-09-03:
`core-image-full-cmdline` redefines `IMAGE_INSTALL` to a minimum setup without ending up pulling `MACHINE_EXTRA_RRECOMMENDS` which in turn includes the WiFi firmware. In the case of the default setup (e.g. using `core-image-base`), this is done through `packagegroup-base-extended`.

closed 2021-09-03 by agherzan

MichaelBMiner, 2022-01-31:
Hello @agherzan I read your two posts about this and I am still confused. I am currently building a core-image-full-cmdline image. I will eventually need Wi-Fi.  Am I able to add the `packagegroup-base-extended` as an `EXTRA_IMAGE_FEATURES` in my local.conf?

Or do I need to add MACHINE_EXTRA_RDEPENDS and MACHINE_EXTRA_RRECOMMENDS sections?

agherzan, 2022-02-02:
If I remember correctly, `packagegroup-base-extended` was pulling `packagegroup-base` which was pulling `packagegroup-machine-base` which was pulling `MACHINE_EXTRA_RRECOMMENDS`. Nevertheless, you can just add `MACHINE_EXTRA_RRECOMMENDS` to `IMAGE_INSTALL` and you should be fine.

gitgonewithwind, 2022-02-02:
@MichaelBMiner 
for me, I used core-image-base to build it ,did not use core-image-full-cmdline