agherzan/meta-raspberrypi #910: raspberrypi4-64 kernel fails to build

dt-nam, 2021-09-09:
**Description**

I think there's a Hosting platform issue with a build dependency. 
Building with the  bblayer (below) and local for the raspberrypi4-64 build fails to build resulting in the following error.

What would cause this?

**Steps to reproduce the issue:**
1. checkout repos
2. init poky
3. bitbake core-image-minimal

**Describe the results you received:**
Build fails
`
| In file included from /home/cyber/prj/yocto/rpi4-64/tmp/work-shared/raspberrypi4-64/kernel-source/arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm2711.dtsi:2:0,
|                  from /home/cyber/prj/yocto/rpi4-64/tmp/work-shared/raspberrypi4-64/kernel-source/arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm2711-rpi-4-b.dts:3,
|                  from /home/cyber/prj/yocto/rpi4-64/tmp/work-shared/raspberrypi4-64/kernel-source/arch/arm64/boot/dts/broadcom/bcm2711-rpi-4-b.dts:1:
| /home/cyber/prj/yocto/rpi4-64/tmp/work-shared/raspberrypi4-64/kernel-source/arch/arm64/boot/dts/broadcom/../../../../arm/boot/dts/bcm283x.dtsi:1:10: fatal error: dt-bindings/pinctrl/bcm2835.h: No such file or directory
|  #include <dt-bindings/pinctrl/bcm2835.h>
|           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
`

**Describe the results you expected:**
Build success

**Additional information you deem important (e.g. issue happens only occasionally):**
Happens with any raspberrypiX-64 build for any release tag. I've tried dunfell, zeus, and hardknott.
**Additional details (revisions used, host distro, etc.):**
Host distro:
DISTRIB_DESCRIPTION="Ubuntu 18.04.5 LTS"

bblayers
 \# POKY_BBLAYERS_CONF_VERSION is increased each time 
 build/conf/bblayers.conf
 \# changes incompatibly
 POKY_BBLAYERS_CONF_VERSION = "2"

 BBPATH = "${TOPDIR}"
 BBFILES ?= ""
 YVER="hardknott"

 BBLAYERS ?= " \
   ${TOPDIR}/../sources/${YVER}/poky/meta \
   ${TOPDIR}/../sources/${YVER}/poky/meta-poky \
   ${TOPDIR}/../sources/${YVER}/poky/meta-yocto-bsp \
   ${TOPDIR}/../sources/${YVER}/meta-raspberrypi \
   ${TOPDIR}/../sources/${YVER}/meta-openembedded/meta-oe \
   ${TOPDIR}/../sources/${YVER}/meta-openembedded/meta-networking \
   ${TOPDIR}/../sources/${YVER}/meta-openembedded/meta-filesystems \
   ${TOPDIR}/../sources/${YVER}/meta-openembedded/meta-python \
   "
local.conf only differs from the default for:
MACHINE ??= "raspberrypi4-64"

kraj, 2021-09-09:
Does it also happen on master ?

dt-nam, 2021-09-09:
I tried with all layers to master and It did not enjoy the layer version mismatch. I could bring only meta-raspberrypi to master. Right now I'm running a new build on a clean docker container of ubuntu:bionic-20210827.

agherzan, 2021-09-10:
What makes you think it is a host platform issue?

dt-nam, 2021-09-10:
As all of the pi 64 builds failed in the same place with clean checkouts of multiple versions I could only guess that there is a build host issue. The clean docker build resulted in an image.

agherzan, 2021-09-10:
So you suspect that somehow this is due to Ubuntu's version?

dt-nam, 2021-09-10:
I suspect it is either a HOST environment variable, Python, or other executable not of the correct definition/version for what the kernel recipe needs done. Not sure what to look at.

agherzan, 2021-09-10:
I'll try to reproduce in an Ubuntu 18.04. So it is not reproducible on 20.04?

dt-nam, 2021-09-10:
The initial build was attempted on a Ubuntu 18.04 box that's been used for development for about 4 years. There would have been changes for different toolchains over time. The docker build kinda proved it to me. I think a clean Ubuntu build would not show this issue, It's probably unique to the development machine I was using.

dt-nam, 2021-09-10:
Just to be clear, building with a clean ubuntu 18.04 docker image resulted in a working raspberrypi4-64 image.

agherzan, 2021-09-10:
I see. In that case I would keep an eye on it and see if you figure out any information that might help us reproduce and issue. I'm proposing to close this issue in the meanwhile.

dt-nam, 2021-09-10:
That’s fine, I’m stumped as to where to look, but I have a feeling it’s related to the python build environment on the failing host. Probably some environmental/packaging cruft left from other tools that messes with the DTC build.

closed 2021-09-10 by dt-nam

kraj, 2021-09-10:
@dt-nam have you installled dtc and libfdt package on your build host by any chance ? if so try to uninstall it.