agherzan/meta-raspberrypi #913: can not correctly add spi rtc m41t93 chip to raspberry pi 4 64bit.

gitgonewithwind, 2021-09-21:
Hi all,
 I want to add a spi rtc chip m41t93 to pi 4 64-bit  , I change the kernel config  to add CONFIG_RTC_DRV_M41T93=y to make sure the kernel driver is on.
I make sure my rtc hardware board works well, I added it to Maaxboard  before, it worked well ,see here: https://community.nxp.com/t5/i-MX-Processors/how-to-make-spi-interface-rtc-function-work-for-imx8mq-maaxboard/m-p/1168938
so similarly, I need to change dtb  in pi 4 to make it work, but after I added the blow patch(which is almost same as that I added in maaxboard), it does not work in pi 4.
any help?
Thank you very much.

how can I do it?

the debug info:
date --set="20201016 20:56:30"                         
Fri Oct 16 20:56:30 UTC 2020                                                   
root@raspberrypi4-64:~# hwclock --systohc --utc --rtc=/dev/rtc0   
root@raspberrypi4-64:~# dmesg | grep rtc
[    3.451357] rtc-m41t93 spi0.0: OF bit is set, write time to restart.
[    3.458578] rtc-m41t93 spi0.0: rtc core: registered rtc-m41t93 as rtc0

dmesg | grep spi
[    3.472773] rtc-m41t93 spi0.0: rtc core: registered rtc-m41t93 as rtc0
root@raspberrypi4-64:~# dmesg | grep rtc
[    3.472773] rtc-m41t93 spi0.0: rtc core: registered rtc-m41t93 as rtc0
root@raspberrypi4-64:~# dmesg | grep spi
[    3.472773] rtc-m41t93 spi0.0: rtc core: registered rtc-m41t93 as rtc0
root@raspberrypi4-64:~# uname -a
Linux raspberrypi4-64 4.19.66 #1 SMP PREEMPT Tue Sep 21 14:41:35 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux

my patch:

cat 0002-rtc-m41t93.patch 
diff -Naur ori/arch/arm/boot/dts/bcm2711-rpi-4-b.dts patch/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
--- ori/arch/arm/boot/dts/bcm2711-rpi-4-b.dts	2021-09-20 11:06:56.930207829 -0400
+++ patch/arch/arm/boot/dts/bcm2711-rpi-4-b.dts	2021-09-21 10:28:02.586670763 -0400
@@ -83,22 +83,15 @@
 	pinctrl-names = "default";
 	pinctrl-0 = <&spi0_pins &spi0_cs_pins>;
 	cs-gpios = <&gpio 8 1>, <&gpio 7 1>;
+	status = "okay";
+            rtc@0 {
+                compatible = "rtc-m41t93";
+                reg = <0>;
+                #address-cells = <1>;
+                #size-cells = <0>;
+                spi-max-frequency = <5000000>;
+            };
 
-	spidev0: spidev@0{
-		compatible = "spidev";
-		reg = <0>;	/* CE0 */
-		#address-cells = <1>;
-		#size-cells = <0>;
-		spi-max-frequency = <125000000>;
-	};
-
-	spidev1: spidev@1{
-		compatible = "spidev";
-		reg = <1>;	/* CE1 */
-		#address-cells = <1>;
-		#size-cells = <0>;
-		spi-max-frequency = <125000000>;
-	};
 };
 
 // =============================================

----

my patch end.

I find someone used the following  in raspbian image(not yocto image), but I do not know how to use it and add it where, I list the patch here too:

/dts-v1/;

/plugin/;

 

/ {

    compatible = "brcm,bcm2835";

    fragment@0 {

        target = <&spi0>;

        __overlay__ {

            spidev@0 {

                status = "disabled";

            };

            spidev@1 {

                status = "disabled";

            };

            rtc-m41t93@0 {

                compatible = "rtc-m41t93";

                reg = <0>;

                #address-cells = <1>;

                #size-cells = <0>;

                spi-max-frequency = <5000000>;

            };

        };

    };

};

gitgonewithwind, 2021-09-21:
hi all,
I added a little bit more  that my upper post ad following :  compatible = "spidev,rtc-m41t93";    (at my first post ,there is no spidev), please see my new attched   file: 002-rtc-m41t93.patch-spidev.txt,I remove the orignal  &spi0)

now after patched is the following,
I run commanad  "hwclock --systohc --utc" and use the scope to capture the wave as the attached files. It seems the cs, clock, miso, mosi are OK,but after power off/on , the date/time is still wrong

any help?
[0002-rtc-m41t93.patch-spidev.txt](https://github.com/agherzan/meta-raspberrypi/files/7205556/0002-rtc-m41t93.patch-spidev.txt)
[image]
[image]
[image]
[image]
[image]
[image]

my patch, I remove spidev0 spidev1, now it is the following:
----
&spi0 {
	pinctrl-names = "default";
	pinctrl-0 = <&spi0_pins &spi0_cs_pins>;
	cs-gpios = <&gpio 8 1>, <&gpio 7 1>;
	status = "okay";
            rtc@0 {
                compatible = "spidev,rtc-m41t93";
                reg = <0>;
                #address-cells = <1>;
                #size-cells = <0>;
                spi-max-frequency = <5000000>;
            };

};

gitgonewithwind, 2021-09-22:
Hi all, I check again, I think my rtc chip is working, set rtc command:"hwclock --systohc --utc".   and read rtc command "hwclock  -r"  work well, but normally, when I use command: "date"  to show it, it always shows my kernel compiling time, so the error is in kernel config or in linux system  config somewhere?
any help?

gitgonewithwind, 2021-09-22:
hi all,
it seems I solve the issue, I  add a command line  "/sbin/hwclock --hctosys " in my start up scripts, it can read rtc time to linux system time. but I did not do it when I add the rtc chip in maaxboard. it seems the linux system get the rtc time from rtc chip automatically in kernel or somewhere?
any help?
Do I have to add the coomand "/sbin/hwclock --hctosys " in startup script?

gitgonewithwind, 2021-09-22:
HI all,
I find the issue:
need to add config in kernel 

CONFIG_RTC_HCTOSYS=y
CONFIG_RTC_HCTOSYS_DEVICE="rtc0"
I am testing...

gitgonewithwind, 2021-09-23:
hi all, 
it's wired.
I have added 
CONFIG_RTC_HCTOSYS=y
CONFIG_RTC_HCTOSYS_DEVICE="rtc0"
in kernel config
and in kernel settings it shows:[*]   Set system time from RTC on startup and resume

but after startup, the date time is still my compilation time.
I have to add "/sbin/hwclock --hctosys " in startup script.
after add  "/sbin/hwclock --hctosys ", the date shows correctly.
any help?

kraj, 2021-09-23:
if you are using sysvinit then you might want to add `rtc` to `MACHINE_FEATURES`

gitgonewithwind, 2021-09-23:
> if you are using sysvinit then you might want to add `rtc` to `MACHINE_FEATURES`

@kraj 
let try to add MACHINE_FEATURES_append = " rtc " in local.conf (right??)

I searched online "RTC and System clock not in sync"  ,find someone mentioned  to use  "/sbin/hwclock --hctosys"  in init.d .
but before I use maaxboard,did not do anything.( I changed pi 4 kernel config same as maaxboard )

kraj, 2021-09-23:
> > if you are using sysvinit then you might want to add `rtc` to `MACHINE_FEATURES`
> 
> @kraj
> let try to add MACHINE_FEATURES_append = " rtc " in local.conf (right??)
> 

yes

> I searched online "RTC and System clock not in sync" ,find someone mentioned to use "/sbin/hwclock --hctosys" in init.d .
> but before I use maaxboard,did not do anything.( I changed pi 4 kernel config same as maaxboard )

you need to include the hwclock init scripts and thats what above option will do

gitgonewithwind, 2021-09-23:
> yes
>
>
> you need to include the hwclock init scripts and thats what above option will do

got it. let me do and test. thanks

gitgonewithwind, 2021-09-23:
@kraj 
sorry, after adding MACHINE_FEATURES_append = " rtc " , the time always is my compilation time.
after I added(or not added) the "/sbin/hwclock --hctosys" in my  startup script  , the time are compilation time.
let me try more tomarrow. 

Thu Sep 23 18:20:55 UTC 2021                                                    
root@raspberrypi4-64:~# hwclock -r                                              
2021-09-23 18:21:03.014401+00:00                                                
root@raspberrypi4-64:~# hwclock -r                                              
2021-09-23 18:21:11.621242+00:00                                                
root@raspberrypi4-64:~# hwclock -r                                              
2021-09-23 18:21:25.460674+00:00                                                
root@raspberrypi4-64:~# hwclock -r                                              
2021-09-23 18:21:27.364927+00:00                                                
root@raspberrypi4-64:~# hwclock -r                                              
2021-09-23 18:21:30.488386+00:00                                                
root@raspberrypi4-64:~# hwclock -r                                              
2021-09-23 18:21:33.428443+00:00

gitgonewithwind, 2021-09-24:
@kraj  and all
after reading /etc/init.d/hwclock.sh , I  add IMAGE_INSTALL_append = "tzdata" in local.conf and ln -s /usr/share/zoneinfo/America/Toronto /etc/localtime
I make it works well.

Thank you very much.
please close this issue.

kraj, 2021-09-24:
thanks closing

closed 2021-09-24 by kraj