agherzan/meta-raspberrypi #914: No sound with DAC HAT

lesti, 2021-09-26:
**Description**

When building an image with the most current version of all used layers (had to update because of the switch to honister anyway) there seems to be a problem with the audio output via the HiFiBerry DAC zero.

**Steps to reproduce the issue:**
1. build an image with `dtoverlay=hifiberry-dac` and `force_eeprom_read=0` in the `RPI_EXTRA_CONFIG` variable
2. boot the system and try to play a wav file (e.g. `aplay file.wav`)

**Describe the results you received:**
There is no sound coming out of the RCA jacks on the DAC when playing the wav file via the DAC.
However, on current RaspberryPi OS (kernel 5.10.17-v7+) with the same hardware configuration, the same wav file plays just fine and the audio comes out of the jacks.

**Describe the results you expected:**
The audio should be audible also on the yocto build. 

**Additional information you deem important (e.g. issue happens only occasionally):**
`aplay -l`/`aplay -L` list the sound card as normal on my distro (looks the same on RaspberryPi OS). However, when running an `alsactl init`, it throws the error of not being able to import an use case config (`Unable to find the top-level configuration file '/usr/share/alsa/ucm2/ucm.conf'` + `error: failed to import hw:0 use case configuration -2`), though I'm not sure if this is important at all.
The card is _not_ muted (not possible without digital mixer in between).

**Additional details (revisions used, host distro, etc.):**
Most current versions of all used layers as of time writing.
meta-raspberrypi @ `9eb4879cf4a289607ec7493577adb0ba97367821`
poky @ `10e73107307ab5ece9d0430af17f3d9f28f457ad`
meta-oe @ `aee2f10638f1407e874970f540c2427a6dc8e6dc`

agherzan, 2021-11-14:
@lesti Sadly I don't have such a hat at my disposal. I would start by investigating this on the Raspbian side, and if things look valid on that side, try to find the differences.

lesti, 2022-10-13:
It seems some recent update fixed this, so I guess we can close this.

closed 2022-10-13 by lesti

agherzan, 2022-10-13:
> It seems some recent update fixed this, so I guess we can close this.

Thanks for confirming.