agherzan/meta-raspberrypi #915: how to install latex in raspberry pi 4 yocto?

gitgonewithwind, 2021-09-28:
hi all, 
my colleague needs latex. I have not idea about it.
is it texlive?
but I try to find it in  the yocto source code,I do not find the bb file about texlive or latex.
where is it? how to find it and add in local.conf？
but my colleague told me in raspbian image, he installed the following packages：
any help?
Thank you very much.

---
nstall Latex on Raspbian:

 

export LC_CTYPE="en_GB.UTF-8"

export LC_ALL="en_GB.UTF-8"

export LANGUAGE="en_GB.UTF-8"

sudo dpkg-reconfigure locales

 

sudo apt-get update

sudo apt-get install texlive-base

sudo apt-get install texlive-latex-recommended

sudo apt-get install texlive

sudo apt-get install texlive-latex-extra

sudo apt-get install texlive-lang-cjk

sudo apt-get install texlive-lang-english

sudo apt-get install texlive-lang-european

sudo apt-get install texlive-lang-spanish

sudo apt-get install texlive-fonts-extra

gitgonewithwind, 2021-09-29:
Anybody did before? can not search the solution online?
Can I use "apt download"  the texlive  (eg: texlive .deb) in raspbian image and use ar xf to uncompress the deb files, and manually install it in my yocto image?

Thanks

agherzan, 2021-10-30:
I have personally never did this and I haven't heard anyone around doing it. I reckon you'd have a better luck to ask people in oe-core or meta-oe as this won't be Raspberry Pi specific.

closed 2021-10-30 by agherzan