agherzan/meta-raspberrypi #929: Support for Raspberry Pi Zero 2 W?

ckelloug3, 2021-10-30:
Are you planning on supporting the Raspberry PI Zero 2 W in Hardknott or later?

kraj, 2021-10-30:
Mostly in master and honister for sure and perhaps dunfell

agherzan, 2021-10-30:
Yes. I can't provide you with an ETA as i don't have a board but I could add the initial support without it. I'll try to look into it soon.

floion, 2021-11-01:
Hi. Do you have an idea if you will create a new machine for it or it will be supposed to work with an updated existing machine in this BSP layer?
Thanks

agherzan, 2021-11-01:
@floion do you have a board at hand? There will be at least a dts addition needed: bcm2710-rpi-zero-2.dts

agherzan, 2021-11-01:
There are some changes to the wireless firmware as well. I'll get a board and add support.

floion, 2021-11-05:
@agherzan yes, we have a board. We'll probably submit a PR with a new machine for it, in `dunfell`, `honister`, and `master`,

floion, 2021-11-05:
What should the approach for this be like? I assume you'd want 2 new machines for both 32 and 64 bits? If so, they'd be inheriting `raspberrypi3.conf` and `raspberrypi3-64.conf` respectively?

kraj, 2021-11-05:
I wonder, if we should just have 64bit only.

floion, 2021-11-07:
Yeah, on our side we'll be using only the 64 bits machine anyway.

acostach, 2021-11-08:
@kraj I raised a PR to add the machines, please let me know if it needs other changes or if I should drop the 32bit one. cc @floion

agherzan, 2021-11-13:
Support merged as part of https://github.com/agherzan/meta-raspberrypi/pull/937

closed 2021-11-13 by agherzan