agherzan/meta-raspberrypi #943: fetching problem for "linux-raspberrypi: Bump to 5.10.78"

mfide, 2021-11-15:
```
bitbake core-image-minimal

Build Configuration:
BB_VERSION           = "1.53.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"
DISTRO               = "poky"
DISTRO_VERSION       = "3.4+snapshot-4aef8c214ba4d5f317d9c47afdfe3d4b7e3c7650"
TUNE_FEATURES        = "aarch64 armv8a crc crypto cortexa72"
TARGET_FPU           = ""
meta
meta-poky
meta-yocto-bsp       = "master:4aef8c214ba4d5f317d9c47afdfe3d4b7e3c7650"
meta-raspberrypi     = "master:c0dd1c9129076d89263aa67a312c5c3de68a67ab"
meta-oe
meta-python
meta-networking
meta-multimedia      = "master:e0ab236378f8a734b72e3b392c900540bd41f87e"

1: linux-raspberrypi-1_5.10.78+gitAUTOINC+e1979ceb17_b2c047ab7e-r0 do_fetch (pid 7498)  76% |###################
```
fetching linux-raspberrypi-1_5.10.78 stops always at 76% and bitbake gives following error:

```
WARNING: linux-raspberrypi-1_5.10.78+gitAUTOINC+e1979ceb17_b2c047ab7e-r0 do_fetch: Failed to fetch URL git://github.com/raspberrypi/linux.git;name=machine;branch=rpi-5.10.y;protocol=https, attempting MIRRORS if available
ERROR: linux-raspberrypi-1_5.10.78+gitAUTOINC+e1979ceb17_b2c047ab7e-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export PATH=...
```

agherzan, 2021-11-15:
We've had very similar reports in the past and they all came down to network issues. Are you able to manually clone the said repository?

mfide, 2021-11-15:
> We've had very similar reports in the past and they all came down to network issues. Are you able to manually clone the said repository?

No this is not a network problem. Reverting the bump mentioned in the title back to linux-raspberrypi-1_5.10.**76** instead of **78** works fine.

agherzan, 2021-11-15:
Could you attach the entire fetch log file?

mfide, 2021-11-15:
> Could you attach the entire fetch log file?

`bitbake core-image-minimal --runall=fetch` command completed successfully. Whole fetch log can be found here:
[20211115122820.log](https://github.com/agherzan/meta-raspberrypi/files/7538745/20211115122820.log)

mfide, 2021-11-15:
Hi, somehow fetch issue disappeared and it became a `NOTE: recipe openssl-3.0.0-r0: task do_package_write_rpm: Failed` issue. I think that the problem is poky related. I will dig further and close this one. Sorry to bother you and Thank you.

closed 2021-11-15 by mfide