agherzan/meta-raspberrypi #944: EXTRA_USERS_PARAMS  not work in yocto update syntax?

gitgonewithwind, 2021-11-15:
hi all,
before I used meta-raspberrypi   commit b112816e95ec59447a82a3a0ee7842c547dda335,
I added the following in local.conf:
NHERIT += "extrausers"
ROOT_PASSWORD = "12345"
EXTRA_USERS_PARAMS = "usermod -p ${ROOT_PASSWORD} root;"
EXTRA_USERS_PARAMS += "useradd -P ${ROOT_PASSWORD} myuser;"
EXTRA_USERS_PARAMS += "usermod -G video myuser;"
I worked well.

but now I use meta-raspberrypi   commit 
commit e31fba4c96cdf819958d05875fb8733481997806 (HEAD -> master, origin/master, origin/HEAD)
Author: Marcel Hamer <marcel@solidxs.se>
Date:   Fri Nov 5 12:18:39 2021 +0100

and I find the syntax of  yocto is updated and I changed something and the lines I added in the local.conf ,but it does not work!

How can I change the lines to make it work? where I can find the new syntax rules?
Thank you very much.

the error shows:
ERROR: core-image-base-1.0-r0 do_rootfs: core-image-base: usermod command did not succeed.
ERROR: core-image-base-1.0-r0 do_rootfs: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs(d)
     0003:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/classes/image.bbclass', lineno: 250, function: do_rootfs
     0246:    progress_reporter.next_stage()
     0247:
     0248:    # generate rootfs
     0249:    d.setVarFlag('REPRODUCIBLE_TIMESTAMP_ROOTFS', 'export', '1')
 *** 0250:    create_rootfs(d, progress_reporter=progress_reporter, logcatcher=logcatcher)
     0251:
     0252:    progress_reporter.finish()
     0253:}
     0254:do_rootfs[dirs] = "${TOPDIR}"
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/lib/oe/rootfs.py', lineno: 383, function: create_rootfs
     0379:
     0380:    img_type = d.getVar('IMAGE_PKGTYPE')
     0381:
     0382:    cls = get_class_for_type(img_type)
 *** 0383:    cls(d, manifest_dir, progress_reporter, logcatcher).create()
     0384:    os.environ.clear()
     0385:    os.environ.update(env_bkp)
     0386:
     0387:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/lib/oe/rootfs.py', lineno: 214, function: create
     0210:        execute_pre_post_process(self.d, rootfs_post_install_cmds)
     0211:
     0212:        self.pm.run_intercepts()
     0213:
 *** 0214:        execute_pre_post_process(self.d, post_process_cmds)
     0215:
     0216:        if self.progress_reporter:
     0217:            self.progress_reporter.next_stage()
     0218:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/lib/oe/utils.py', lineno: 263, function: execute_pre_post_process
     0259:    for cmd in cmds.strip().split(';'):
     0260:        cmd = cmd.strip()
     0261:        if cmd != '':
     0262:            bb.note("Executing %s ..." % cmd)
 *** 0263:            bb.build.exec_func(cmd, d)
     0264:
     0265:# For each item in items, call the function 'target' with item as the first
     0266:# argument, extraargs as the other arguments and handle any exceptions in the
     0267:# parent thread
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/bitbake/lib/bb/build.py', lineno: 256, function: exec_func
     0252:    with bb.utils.fileslocked(lockfiles):
     0253:        if ispython:
     0254:            exec_func_python(func, d, runfile, cwd=adir)
     0255:        else:
 *** 0256:            exec_func_shell(func, d, runfile, cwd=adir)
     0257:
     0258:    try:
     0259:        curcwd = os.getcwd()
     0260:    except:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/bitbake/lib/bb/build.py', lineno: 507, function: exec_func_shell
     0503:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0504:        try:
     0505:            bb.debug(2, "Executing shell function %s" % func)
     0506:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0507:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0508:        except bb.process.ExecutionError as exe:
     0509:            # Find the backtrace that the shell trap generated
     0510:            backtrace_marker_regex = re.compile(r"WARNING: Backtrace \(BB generated script\)")
     0511:            stdout_lines = (exe.stdout or "").split("\n")
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/bitbake/lib/bb/process.py', lineno: 186, function: run
     0182:
     0183:    if pipe.returncode != 0:
     0184:        if log:
     0185:            # Don't duplicate the output in the exception if logging it
 *** 0186:            raise ExecutionError(cmd, pipe.returncode, None, None)
     0187:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0188:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/home/rnd2/work/pi4/nacho-update/build/tmp/work/raspberrypi4_64-poky-linux/core-image-base/1.0-r0/temp/run.set_user_group.11128' failed with exit code 1

ERROR: Logfile of failure stored in: /home/rnd2/work/pi4/nacho-update/build/tmp/work/raspberrypi4_64-poky-linux/core-image-base/1.0-r0/temp/log.do_rootfs.11128
ERROR: Task (/home/rnd2/work/pi4/nacho-update/layers/poky/meta/recipes-core/images/core-image-base.bb:do_rootfs) failed with exit code '1'

agherzan, 2021-11-16:
Nothing in your report strikes me as a change in yocto syntax. There is a typo there `NHERIT` but assuming that it is a copy paste issue. That being said, there are very slim chances for this to be a due to meta-raspberrypi. Have you tried to replicate this issue while removing meta-raspberrypi and building something like a qemux86 machine?

gitgonewithwind, 2021-11-16:
@agherzan 
sorry, NHERIT is a copy paste issue.  there is no the issue in my full local.conf. let me try qemux86 machine

agherzan, 2021-11-16:
I'll close this issue for now - it doesn't sound `meta-raspberrypi` specific but feel free to come back to us with details that show otherwise.

closed 2021-11-16 by agherzan

gitgonewithwind, 2021-12-09:
hi @agherzan 

I tried qemux86 machine , but it can not work either.
any help?

error info:

ERROR: core-image-base-1.0-r0 do_rootfs: core-image-base: useradd command did not succeed.
ERROR: core-image-base-1.0-r0 do_rootfs: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs(d)
     0003:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/classes/image.bbclass', lineno: 250, function: do_rootfs
     0246:    progress_reporter.next_stage()
     0247:
     0248:    # generate rootfs
     0249:    d.setVarFlag('REPRODUCIBLE_TIMESTAMP_ROOTFS', 'export', '1')
 *** 0250:    create_rootfs(d, progress_reporter=progress_reporter, logcatcher=logcatcher)
     0251:
     0252:    progress_reporter.finish()
     0253:}
     0254:do_rootfs[dirs] = "${TOPDIR}"
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/lib/oe/rootfs.py', lineno: 383, function: create_rootfs
     0379:
     0380:    img_type = d.getVar('IMAGE_PKGTYPE')
     0381:
     0382:    cls = get_class_for_type(img_type)
 *** 0383:    cls(d, manifest_dir, progress_reporter, logcatcher).create()
     0384:    os.environ.clear()
     0385:    os.environ.update(env_bkp)
     0386:
     0387:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/lib/oe/rootfs.py', lineno: 214, function: create
     0210:        execute_pre_post_process(self.d, rootfs_post_install_cmds)
     0211:
     0212:        self.pm.run_intercepts()
     0213:
 *** 0214:        execute_pre_post_process(self.d, post_process_cmds)
     0215:
     0216:        if self.progress_reporter:
     0217:            self.progress_reporter.next_stage()
     0218:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/meta/lib/oe/utils.py', lineno: 263, function: execute_pre_post_process
     0259:    for cmd in cmds.strip().split(';'):
     0260:        cmd = cmd.strip()
     0261:        if cmd != '':
     0262:            bb.note("Executing %s ..." % cmd)
 *** 0263:            bb.build.exec_func(cmd, d)
     0264:
     0265:# For each item in items, call the function 'target' with item as the first
     0266:# argument, extraargs as the other arguments and handle any exceptions in the
     0267:# parent thread
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/bitbake/lib/bb/build.py', lineno: 256, function: exec_func
     0252:    with bb.utils.fileslocked(lockfiles):
     0253:        if ispython:
     0254:            exec_func_python(func, d, runfile, cwd=adir)
     0255:        else:
 *** 0256:            exec_func_shell(func, d, runfile, cwd=adir)
     0257:
     0258:    try:
     0259:        curcwd = os.getcwd()
     0260:    except:
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/bitbake/lib/bb/build.py', lineno: 507, function: exec_func_shell
     0503:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0504:        try:
     0505:            bb.debug(2, "Executing shell function %s" % func)
     0506:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0507:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0508:        except bb.process.ExecutionError as exe:
     0509:            # Find the backtrace that the shell trap generated
     0510:            backtrace_marker_regex = re.compile(r"WARNING: Backtrace \(BB generated script\)")
     0511:            stdout_lines = (exe.stdout or "").split("\n")
File: '/home/rnd2/work/pi4/nacho-update/layers/poky/bitbake/lib/bb/process.py', lineno: 186, function: run
     0182:
     0183:    if pipe.returncode != 0:
     0184:        if log:
     0185:            # Don't duplicate the output in the exception if logging it
 *** 0186:            raise ExecutionError(cmd, pipe.returncode, None, None)
     0187:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0188:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/home/rnd2/work/pi4/nacho-update/build/tmp/work/qemux86_64-poky-linux/core-image-base/1.0-r0/temp/run.set_user_group.25283' failed with exit code 1

ERROR: Logfile of failure stored in: /home/rnd2/work/pi4/nacho-update/build/tmp/work/qemux86_64-poky-linux/core-image-base/1.0-r0/temp/log.do_rootfs.25283
ERROR: Task (/home/rnd2/work/pi4/nacho-update/layers/poky/meta/recipes-core/images/core-image-base.bb:do_rootfs) failed with exit code '1'

my full local.conf file for  qemux86-64:

MACHINE ??= "qemux86-64"

DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_rpm"

EXTRA_IMAGE_FEATURES ?= "debug-tweaks"

USER_CLASSES ?= "buildstats"

PATCHRESOLVE = "noop"
INHERIT += "extrausers"
ROOT_PASSWORD = "12345"
EXTRA_USERS_PARAMS = "usermod -p ${ROOT_PASSWORD} root;"
EXTRA_USERS_PARAMS += "useradd -P ${ROOT_PASSWORD} myuser;"
BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

PACKAGECONFIG:append:pn-qemu-system-native = " sdl"
CONF_VERSION = "2"

gitgonewithwind, 2021-12-10:
@agherzan 
can you open this issue again? thank you

agherzan, 2022-01-04:
@gitgonewithwind I am not happy to open it in meta-raspberrypi unless we have a good reason to believe that this functionality is broken when using this layer. Have you tried this oe-core alone and report the issue in Bugzilla or on the oe-core mailing list?

weiqj, 2022-01-15:
I too had this problem. Most likely it is a Yocto issue.
I used EXTRA_IMAGE_FEATURES = "debug-tweaks" instead.
And I can testify that debug-tweaks works well.

gitgonewithwind, 2022-01-16:
@weiqj 
thanks, I am going to try.

gitgonewithwind, 2022-02-04:
@weiqj 
it's strange. not work for me EXTRA_IMAGE_FEATURES = "debug-tweaks"
the error is still there:
anybody any help?

RROR: core-image-base-1.0-r0 do_rootfs: core-image-base: useradd command did not succeed.
ERROR: core-image-base-1.0-r0 do_rootfs: Error executing a python function in exec_func_python() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_func_python() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:do_rootfs(d)
     0003:
File: '/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/meta/classes/image.bbclass', lineno: 251, function: do_rootfs
     0247:    progress_reporter.next_stage()
     0248:
     0249:    # generate rootfs
     0250:    d.setVarFlag('REPRODUCIBLE_TIMESTAMP_ROOTFS', 'export', '1')
 *** 0251:    create_rootfs(d, progress_reporter=progress_reporter, logcatcher=logcatcher)
     0252:
     0253:    progress_reporter.finish()
     0254:}
     0255:do_rootfs[dirs] = "${TOPDIR}"
File: '/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/meta/lib/oe/rootfs.py', lineno: 383, function: create_rootfs
     0379:
     0380:    img_type = d.getVar('IMAGE_PKGTYPE')
     0381:
     0382:    cls = get_class_for_type(img_type)
 *** 0383:    cls(d, manifest_dir, progress_reporter, logcatcher).create()
     0384:    os.environ.clear()
     0385:    os.environ.update(env_bkp)
     0386:
     0387:
File: '/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/meta/lib/oe/rootfs.py', lineno: 214, function: create
     0210:        execute_pre_post_process(self.d, rootfs_post_install_cmds)
     0211:
     0212:        self.pm.run_intercepts()
     0213:
 *** 0214:        execute_pre_post_process(self.d, post_process_cmds)
     0215:
     0216:        if self.progress_reporter:
     0217:            self.progress_reporter.next_stage()
     0218:
File: '/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/meta/lib/oe/utils.py', lineno: 263, function: execute_pre_post_process
     0259:    for cmd in cmds.strip().split(';'):
     0260:        cmd = cmd.strip()
     0261:        if cmd != '':
     0262:            bb.note("Executing %s ..." % cmd)
 *** 0263:            bb.build.exec_func(cmd, d)
     0264:
     0265:# For each item in items, call the function 'target' with item as the first
     0266:# argument, extraargs as the other arguments and handle any exceptions in the
     0267:# parent thread
File: '/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/bitbake/lib/bb/build.py', lineno: 256, function: exec_func
     0252:    with bb.utils.fileslocked(lockfiles):
     0253:        if ispython:
     0254:            exec_func_python(func, d, runfile, cwd=adir)
     0255:        else:
 *** 0256:            exec_func_shell(func, d, runfile, cwd=adir)
     0257:
     0258:    try:
     0259:        curcwd = os.getcwd()
     0260:    except:
File: '/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/bitbake/lib/bb/build.py', lineno: 507, function: exec_func_shell
     0503:    with open(fifopath, 'r+b', buffering=0) as fifo:
     0504:        try:
     0505:            bb.debug(2, "Executing shell function %s" % func)
     0506:            with open(os.devnull, 'r+') as stdin, logfile:
 *** 0507:                bb.process.run(cmd, shell=False, stdin=stdin, log=logfile, extrafiles=[(fifo,readfifo)])
     0508:        except bb.process.ExecutionError as exe:
     0509:            # Find the backtrace that the shell trap generated
     0510:            backtrace_marker_regex = re.compile(r"WARNING: Backtrace \(BB generated script\)")
     0511:            stdout_lines = (exe.stdout or "").split("\n")
File: '/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/bitbake/lib/bb/process.py', lineno: 186, function: run
     0182:
     0183:    if pipe.returncode != 0:
     0184:        if log:
     0185:            # Don't duplicate the output in the exception if logging it
 *** 0186:            raise ExecutionError(cmd, pipe.returncode, None, None)
     0187:        raise ExecutionError(cmd, pipe.returncode, stdout, stderr)
     0188:    return stdout, stderr
Exception: bb.process.ExecutionError: Execution of '/z/oe/pi4-v1.4/touchscreen-1.4/build/tmp/work/raspberrypi4_64-poky-linux/core-image-base/1.0-r0/temp/run.set_user_group.19929' failed with exit code 1

ERROR: Logfile of failure stored in: /z/oe/pi4-v1.4/touchscreen-1.4/build/tmp/work/raspberrypi4_64-poky-linux/core-image-base/1.0-r0/temp/log.do_rootfs.19929
ERROR: Task (/z/oe/pi4-v1.4/touchscreen-1.4/layers/poky/meta/recipes-core/images/core-image-base.bb:do_rootfs) failed with exit code '1'

gitgonewithwind, 2022-02-04:
@weiqj 
hi all, I have found the solutions. that's the updated syntax here:
my poky version is 3.4
https://docs.yoctoproject.org/3.4.1/ref-manual/classes.html?highlight=useradd

example for me:
in local.conf I add the below lines:

INHERIT += "extrausers"

#mkpasswd -m sha256crypt testpassword
ROOT_PASSWORD = "\$5\$XXXXXXX\$XXXXXXXXXXXXXXXXXXXXXXXXXX"
EXTRA_USERS_PARAMS = "usermod -p '${ROOT_PASSWORD}' root; \
                      useradd -p '${ROOT_PASSWORD}' myuser; \
                      usermod -G video myuser; \
"
explain:
1. use mkpasswd -m sha256crypt testpassword   to generate the hash  for the passwaord: testpassword 
as below:
"$5$XXXXXXX$XXXXXXXXXXXXXXXXXXXXXXXXXX"
2. add the escape  "\" for the 3 letter $       in the hash
3. put the added "\"  hash at:  ROOT_PASSWORD = "\$5\$XXXXXXX\$XXXXXXXXXXXXXXXXXXXXXXXXXX"

that's it. make my headache for a long time.
good luck  for new comers!

gitgonewithwind, 2022-02-12:
@agherzan  welcome